tinygs/tinyGS #13: Ground Station Arduino Code for ESP32 and Lora board without the OLED

Cloolalang, 2019-12-09:
I am in South Africa, I am ready to setup my ground station but I dont have a Heltec or TTGO board...
I have an ESP32 Dev Board with an AI thinker RA02 for Lora at 433Mhz it is working and is based in this:

https://randomnerdtutorials.com/esp32-lora-rfm95-transceiver-arduino-ide/

Is there a version of the ground station Arduino firmware without the OLED stuff?

Cloolalang, 2019-12-10:
All Good I just deleted out all the oLED stuff

4m1g0, 2019-12-16:
Hello, thanks for the info.

It could be interesting to add a flag to disable the OLED in configuration for all the boards that do not have it.

lillefyr, 2019-12-17:
It would be nice to have all the configurable parameters in one file.  I use a Lolin D32 v1.0 board,

G4lile0, 2021-02-17:
Fixed on the new release

closed 2021-02-17 by G4lile0