tinygs/tinyGS #15: Add rtty decoding on board with sx1276 (heltec)

lucapgt, 2019-12-10:
Hi,
Can you add rtty decoding on groundstation with sx1276?
Othewise this boards Is useless.

Thanks

4m1g0, 2019-12-11:
Hi,
We are looking at that. However, the main LoRa idea is still active:

- Boards with sx127x will (as Julian said) be compatible with the new satellites launched in March.
- We've added compatibility with SX126x
- There has been success decoding [one package](https://twitter.com/G4lile0/status/1204311425025486848). If the conditions of the satellite improve it might be possible to decode some more even though it's not optimal...

We will use this issue to keep track of new advances in RTTY decoding in case it is possible ;)

G4lile0, 2019-12-12:
Seems that decoding RTTY with SX1278, isn't possible as the radio can't distinguish the mark and space frequencies as they are just 182 Hz apart

closed 2019-12-12 by G4lile0