tinygs/tinyGS #34: Incorrect Tx frame length calculated if data contains null.

PaulSchulz, 2020-02-02:
https://github.com/G4lile0/ESP32-OLED-Fossa-GroundStation/blob/fb1eb6971e3091b8dee7a2edfa90df4f162235d3/FossaGroundStation/src/Radio/Radio.cpp#L135

optDataLen is calculated using 'strlen'. This will give an incorrect value if the data contains a null byte, and will transmit a short packet from the ground station.

I would suggenst passing size of optData. Patch may follow.

4m1g0, 2020-02-09:
When can that happen? data is obtained as a string terminated by null

PaulSchulz, 2020-02-10:
It won't happen in the code at the moment, but I tried to use the code to send emulated satellite system data, and had issues when sending zero values in the middle of a frame.

(I checked the github code for the FOSSASAT-1 satellite and it's also handled differently.) 

I was also thinking of being able send/receive binary data via the serial interface, but that's a longer term project.

4m1g0, 2021-02-06:
This is handled in a total different way now. It should not be an issue anymore.

closed 2021-02-06 by 4m1g0