tinygs/tinyGS #35: TCXO support for sx1262

N6RFM, 2020-02-02:
Fantastic project.  Thanks all support.  Using the sx1262 /B v1.1 chip from Nice-RF.  The radio can be successfully enabled using board: 5 -->  Custom ESP32 Wroom + SX126x (Crystal).  However, this sx1262 chip contains a TCXO.  Simply changing the voltage from 0.0f to 1.6f results in a failure code.  Please crate a new board option or code modification for this chip which enables the TCXO.  Thanks again.

PaulSchulz, 2020-02-03:
Just trying to get some more information.
What needs to be changed? Board 5 has the following parameters (as per 'ConfigManager/ConfigManger.cpp'). What needs to be changed?
  OLED_add: 0x3c
  OLED_SDA: 21
  OLED_SCL: 22
  OLED_RST: 16
  PROG_BUTTON: 0
  BOARD_LED: 25 
  L_SX127X?: false
  L_NSS: 5
  L_DI00: 0
  L_DI01: 27
  L_BUSSY: 26
  L_RST: 14
  L_MISO: 19
  L_MOSI: 23
  L_SCK: 18
  L_TCXO_V: 0.0f
  BOARD: "Custom ESP32 Wroom + SX126x (Crystal)"

PaulSchulz, 2020-02-03:
- Looking in the code, the L_TCXO_V is used if L_SX127X? is false (which it is). (Radio/Radio.cpp:67)

- Chip reference: https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/2R000000HT76/7Nka9W5WgugoZe.xwIHJy6ebj1hW8UJ.USO_Pt2CLLo
see section 4.1.4 TCXO Control Block, 13.3.6 SetDIO3AsTCXOCtl

N6RFM, 2020-02-03:
Hi Paul,

Ideally, setting 0.0f to 1.6f would enable the TCXO.  Unfortunately, that is not working.  The chip developer has provided test code, shared with 4M1GO, in which the TCXO is enabled for further investigation. 
[SX1262_Arduino_Demo.zip](https://github.com/G4lile0/ESP32-OLED-Fossa-GroundStation/files/4149153/SX1262_Arduino_Demo.zip)
To your point, DI03 may also be used to control the TCXO by applying an external voltage.  I tried that.  The chip enables properly, but I am unable to confirm if the code to actuate the switch remains in the sx126x driver in RadioLib.  Also, we need to keep in mind that the RadioLib version we are using here is patched in some manner.  Changing out to stock RadioLib libraries does not work. 
 
Further, it appears that at the moment that if a sx1278 chip is not specified, then the software defaults to an sx1268.  Maybe this is an incorrect interpretation on my part.  Also, not sure if that is actually a factor in TCXO behavior or not.  Unfortunately, changing instances of sx1268 to sx1262 in the Radio.cpp (line 66) and /or adding board.L_TCXO_V to line 66 of Radio.cpp do not resolve the -707 error, when changing the TCXO voltage from 0.0f to 1.6f.  Thus, for reasons I have been unable to identify, it appears that the sx1268 (or sx126x) libraries we are using here do not appear to implement TCXO control for this specific chip.  Oscar, 4M1GO has kindly agreed to look further into this.  Fortunately and in the meantime, the chip does work when the voltage parameter is set to 0.  Hopefully, the root cause can be identified and this chip feature be fully realized, as is already the case for the Dorji DRF1268T chip.  In addition, other than the somewhat bulky Dorji DAD06 testing board for the DRF1268T, I have been unable to find conveniently sized breakouts for the DRF1268T.  

So, the Nice RF sx1262 /B v1.1 chip appeared to be a good alternative as there are several small breakout boards available to help end users like me avoid the micro-soldering of tiny wires otherwise required for practical use.  Others have been successful in such soldering but alas not I.  As an alternative, I have pursued the sx1262.  This software and system works nicely!  But, enabling the TXCO remains attractive, and on my wish list.  Comments, suggestions most welcome!  

Thanks,  Bob

PaulSchulz, 2020-02-06:
Ok, thank you for adding the details. A bit more complicated than what I was expecting. I knew that the groundstation code was using a patched version of RadioLib, but I don't know or understand what those differences are yet.

iw2lsi, 2021-08-24:
Hi,

   any news on tinyGS w/ NiceRF 1262 module ?

I bought an "G-NiceRF VHF 150mhz LoRa1268 SX1268 LoRa RF Module 22dBm 160mW 10ppm SPI Ultra Low consumption" from Aliexpress... which is actually (according to the manufacturer) NOT an 1268 but indeed an 1262 (with no TCXO) for VHF band (150MHz).

I'm using this wiring/template (with an [attexx/rfm9x_breakout_board](https://github.com/attexx/rfm9x_breakout_board)) and the radio seems to be initialized...

board type: "Custom ESP32 Wroom + SX126x (Crystal)" --> an ESP32-DevKitC v4

template:

{"name":"[433] DevC4 NiceRF","aADDR":60,"oSDA":0,"oSCL":0,"oRST":0,"pBut":0,"led":0,"radio":0,"lNSS":18,"lDIO0":26,"lDIO1":33,"lBUSSY":25,"lRST":23,"lMISO":19,"lMOSI":27,"lSCK":5,"lTCXOV":0.0}

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
TinyGS Version 2105260 - 2105260
[SX12xx] Initializing ...
[SX12x8] Starting to listen to Norbi
00:00:01 ------------- Controls -------------
00:00:01 t - change the test mode and restart
00:00:01 e - erase board config and reset
00:00:01 b - reboot the board
00:00:01 p - send test packet to nearby stations (to check transmission)
00:00:01 ------------------------------------

BTW: before mapping the BUSSY pin to GPIO25 I was also getting error -707

Now the question is: how can I check the RTX in VHF ? does tinyGS support it ? 
can I just specify the frequency in the template ?

      Giampaolo

DroneMechanic, 2021-08-24:
If you have a second set you can do the test transmit packet. Then can test the rx, this would be the easiest n fastest way.

iw2lsi, 2021-08-25:
ok... it seems to work now... but I had to hack the source code and replacing the 1268 radio module with 1262... the TX is now confirmed by an SDR dongle @150MHz

In the next days I'll wire a second NiceRF module and test RTX from eachother...

For the official support I think we'll have to extend the radio=n set by adding another value (eg: radio=2 for 1262)

    Giampaolo