tinygs/tinyGS #43: Override Auto-Update to Radio Settings?

K4KDR, 2020-10-15:
Hello!

Is there a "flag" or other means in place to defeat the behavior of the RF Freq (and other LoRa settings) from being reset by the system automatically?

When I manually change the LoRa parameters in` ./src/Radio/Radio.cpp` and upload to my device, if I am understanding the code as-written, the radio settings on my device reset when the MQTT connection is established.

If there is not a method to override this behavior, could one be added, please?  I would like to determine whether the ESP32 (in conjunction with the excellent system you have setup for NORBI/Fossa testing) can receive the 137 MHz SpaceBee LoRa downlinks.    (my '433 MHz' Dragino board on an Arduino Uno receives the SpaceBee signals perfectly!)

Thank you!

-Scott,  K4KDR

G4lile0, 2020-10-17:
Hi Scott, 

Great experiment !!! Adding Spacebee Lora satellites will be really cool. 
Right we are no sending any remote command to control the GS stations, "Norby" is the default sat, if you change the config on Radio.cpp they will not change after the MQTT connection, if you want to change the info on the display, you should change the Status.h file.

In any case, I suggest to keep the original firmware and remote control your own Ground Station, just send a command send these mqtt commands, with your mqtt user and password.

**Change the Remote_Satellite_Name to SpaceBee**
`mosquitto_pub -u YOUR_MQTT_USER -P YOUR_MQTT_PASSWORD --cafile certificate.cer -h fossa.apaluba.com -p 8883 -m  "[\"SpaceBee\"]" -t fossa/1210637909/test_K4KDR/data/remote/sat`

**Change the Frecuency to 137.975 Mhz**
`mosquitto_pub -u YOUR_MQTT_USER -P YOUR_MQTT_PASSWORD --cafile certificate.cer -h fossa.apaluba.com -p 8883 -m "[137.975]" -t fossa/1210637909/test_K4KDR/data/remote/freq`

**Change Bandwith to 41.7 khz**
`mosquitto_pub -u YOUR_MQTT_USER -P YOUR_MQTT_PASSWORD --cafile certificate.cer -h fossa.apaluba.com -p 8883 -m "[41.7]" -t fossa/1210637909/test_K4KDR/data/remote/bw`

**Change spreading factor to 10**
`mosquitto_pub -u YOUR_MQTT_USER -P YOUR_MQTT_PASSWORD --cafile certificate.cer -h fossa.apaluba.com -p 8883 -m "[10]"  -t fossa/1210637909/test_K4KDR/data/remote/`

**Change Coding Rate to 5**
`mosquitto_pub -u YOUR_MQTT_USER -P YOUR_MQTT_PASSWORD --cafile certificate.cer -h fossa.apaluba.com -p 8883 -m "[5]" -t fossa/1210637909/test_K4KDR/data/remote/cr`

**Change Preamble to 16**
`mosquitto_pub -u YOUR_MQTT_USER -P YOUR_MQTT_PASSWORD --cafile certificate.cer -h fossa.apaluba.com -p 8883 -m "[16]" -t fossa/1210637909/test_K4KDR/data/remote/pl`

No need to change SyncWord (no need to do it as Norby use the same sync word)

**You may need to enable or disable CRC:**
```
-m "[1]" -t fossa/1210637909/test_K4KDR/data/remote/crc
-m "[0]" -t fossa/1210637909/test_K4KDR/data/remote/crc

```
**Force or disable LDRO**
```
-m "[1]" -t fossa/1210637909/test_K4KDR/data/remote/fldro
-m "[0]" -t fossa/1210637909/test_K4KDR/data/remote/fldro
```

**Enable AutoLDRO**
`-m "[1]" -t fossa/1210637909/test_K4KDR/data/remote/aldro
`

to create the certicate, copy this test on a txt file

```
-----BEGIN CERTIFICATE-----
MIIDSjCCAjKgAwIBAgIQRK+wgNajJ7qJMDmGLvhAazANBgkqhkiG9w0BAQUFADA/
MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT
DkRTVCBSb290IENBIFgzMB4XDTAwMDkzMDIxMTIxOVoXDTIxMDkzMDE0MDExNVow
PzEkMCIGA1UEChMbRGlnaXRhbCBTaWduYXR1cmUgVHJ1c3QgQ28uMRcwFQYDVQQD
Ew5EU1QgUm9vdCBDQSBYMzCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEB
AN+v6ZdQCINXtMxiZfaQguzH0yxrMMpb7NnDfcdAwRgUi+DoM3ZJKuM/IUmTrE4O
rz5Iy2Xu/NMhD2XSKtkyj4zl93ewEnu1lcCJo6m67XMuegwGMoOifooUMM0RoOEq
OLl5CjH9UL2AZd+3UWODyOKIYepLYYHsUmu5ouJLGiifSKOeDNoJjj4XLh7dIN9b
xiqKqy69cK3FCxolkHRyxXtqqzTWMIn/5WgTe1QLyNau7Fqckh49ZLOMxt+/yUFw
7BZy1SbsOFU5Q9D8/RhcQPGX69Wam40dutolucbY38EVAjqr2m7xPi71XAicPNaD
aeQQmxkqtilX4+U9m5/wAl0CAwEAAaNCMEAwDwYDVR0TAQH/BAUwAwEB/zAOBgNV
HQ8BAf8EBAMCAQYwHQYDVR0OBBYEFMSnsaR7LHH62+FLkHX/xBVghYkQMA0GCSqG
SIb3DQEBBQUAA4IBAQCjGiybFwBcqR7uKGY3Or+Dxz9LwwmglSBd49lZRNI+DT69
ikugdB/OEIKcdBodfpga3csTS7MgROSR6cz8faXbauX+5v3gTt23ADq1cEmv8uXr
AvHRAosZy5Q6XkjEGB5YGV8eAlrwDPGxrancWYaLbumR9YbK+rlmM6pZW87ipxZz
R8srzJmwN0jP41ZL9c8PDHIyh8bwRLtTcm1D9SZImlJnt1ir/md2cXjbDaJWFBM5
JDGFoqgCWjBH4d1QB7wCCZAA62RjYJsWvIjJEubSfZGL+T0yjWW06XyxV3bqxbYo
Ob8VZRzI9neWagqNdwvYkQsEjgfbKbYK7p2CNTUQ
-----END CERTIFICATE-----
```

Good luck and keep us updated !

K4KDR, 2020-10-17:
Thank you VERY much for the detailed reply!

While I might not have EVERY LoRa setting correct yet, I am getting excellent LoRa decodes from the SpaceBee constellation on 137 MHz using an Arduino UNO w/ the 433 MHz Dragino shield.  

https://twitter.com/scott23192/status/1316603189903396864

Unfortunately, with the same settings, my ESP32/HELTEC WiFi LoRa 32 V2 combo does NOT decode any of the SpaceBee 137 MHz LoRa packets.  So, unless I have missed a setting, it would appear that the ESP32 combo that works so well for NORBI is not as efficient outside of the 433 MHz band.

Interestingly, when I program the ESP32 with the 137 MHz SpaceBee settings, I can TRANSMIT on 137 MHz and the payload is both visible on an SDR waterfall (on the expected 137 MHz frequency) and is also DECODED properly by the Arduino UNO / 433 MHz Dragino shield combo.

So, the ESP32 is definitely accepting the settings change.  However, even in close proximity to the Arduino (used as a test TX source), the ESP32 absolutely refuses to decode anything on 137 MHz.

I will continue to experiment.  For now, my 'live' over-the-air 137 MHz reception will have to use the Arduino/Dragino combo.  But I will continue to try various options on the ESP32 in case it can eventually be found to operate outside of the 433 MHz band.  

Finally, I have a pair of  433 MHz TTGO ESP32 boards on order.  So, if those happen to work as well outside of the target 433 MHz band, I will post an  update.

G4lile0, 2020-10-17:
Thanks for the update, 

Be careful with transmission on 137Mhz, if the antenna doesn't have the right impedance you can burn the SX1272 module on the board.
Do you know if the Spacebee use LoRa LDRO mode ? 

 Thanks!

K4KDR, 2020-10-17:
>>> Do you know if the Spacebee use LoRa LDRO mode ?

I do not.  Here are the settings that, while possibly not 100% correct, are producing LoRa decodes from the SpaceBee constellation using my Arduino UNO w/ the 433 MHz Dragino shield:

```
#define FREQUENCY             137.975  // MHz

#define BANDWIDTH             41.7     // Sets LoRa bandwidth. Allowed values are 7.8, 10.4, 15.6, 20.8,
                                           31.25, 41.7, 62.5, 125.0, 250.0 and 500.0 kHz.
                                           
#define SPREADING_FACTOR      8        // Sets LoRa spreading factor. Allowed values range from 5 to 12.

#define CODING_RATE           5        // Sets LoRa coding rate 4/x denominator.
                                           Allowed x values range from 5 to 8.
                                           
#define LORA_PREAMBLE_LEN     16       // preambleLength LoRa preamble length in symbols.
                                           Allowed values range from 1 to 65535.
                                           
#define SYNC_WORD             0x12     // used as LoRa "sync word"
```

N6RFM, 2020-10-24:
Hi @G4lile0 .  How long should it take for mqtt remote control commands to propagate back to the remote esp32?  I've tried to change the SF a few times, but the OLED does not change.  Issuing command seems to work OK.

Thanks,

Bob

4m1g0, 2021-02-17:
This has been modified with the new release. For future reference check:
* https://github.com/G4lile0/tinyGS/wiki/Radio-Tuning-Guide
* https://github.com/G4lile0/tinyGS/wiki/Programmatic-API

closed 2021-02-17 by 4m1g0