tinygs/tinyGS #45: Add support for WebSerial Monitor

fooons, 2020-10-17:
I wonder if this improvement/enhancement could be added to our GroundStations to be able to send commands to the web serial port and receive.  Unfortunately my TTGO crashed and I have lost the USB port.  I upload the software via Arduino and the network port.  I can solder some small cables, but for those of us who have had this breakdown or those who do not have the station nearby, it would be a great advance.

I have seen in GitHub a Remote Serial monitor for ESP8266 & ESP32 and I'm trying to use it, but until new board with USB arrives I'm not sure to program mine with changes.

Thanks in advance

fooons, 2020-11-09:
Better than WebSerial I have used a Telnet server to monitor some messages and use the controls menu to erase and reboot in my home network. It has been a quick implementation so needs polish.
[image]

hjf, 2021-01-16:
have you uploaded this code somewhere?

4m1g0, 2021-02-17:
Thanks for the suggestion.

Added web console on @21021701 

https://github.com/G4lile0/tinyGS/wiki/Local-Dashboard

closed 2021-02-17 by 4m1g0