tinygs/tinyGS #50: RadioLib version

jgromes, 2020-10-24:
Hi guys,

I was trying out your ground station project and came across the following:

https://github.com/G4lile0/ESP32-OLED-Fossa-GroundStation/blob/f2f1af0619a6eed596f3d112e42b64e79a9a739f/FossaGroundStation/FossaGroundStation.ino#L80-L82

This seems unneccessary, since the version of RadioLib in the lib folder is identical to the upstream (as far as I can tell). Moreover, it causes an issue if the user already has RadioLib installed, either cloned or via Arduino library manager (when using Arduino IDE).

Also, maybe it would be easier to keep a git submodule in the lib folder instead of a hard copy (so that you don't have to manually copy-paste changes) - similarly for the other libraries.

4m1g0, 2020-10-25:
Hello jgromes, nice to see you over here.

You are right, currently it does not make sense anymore. This was added because we were temporarily using a modified version when some of the stations were making tests trying to receive FossaSat-1 with the old syncword.

We are currently making a lot of changes on the backend and the firmware to make it more flexible and be able to support features that were not planned at the beginning. We were currently making some fast changes to support Norby sat including the update to the upstream of RadioLib, but the documentation and other segments of the firmware are currently outdated.

We plan to make a big refactor and release a new version in the next couple of weeks, and the RadioLib library will point to the upstream for sure!

jgromes, 2020-10-25:
Thanks - feel free to open issues in RadioLib repo if there's anything that should be changed in the library. IMO it's always better to keep this sort of thing over there.

BTW it's really cool to see this project support more LoRa satellites - sort of like a small LoRa SatNOGS :)

closed 2020-10-25 by jgromes