tinygs/tinyGS #60: New version, with new MQTT login "failed, rc=-2"

zanco, 2021-02-18:
Hi, Today I upgraded 2 times TTGO which both have run the Fossasat version without problems.

Both are running (flashed with Arduino IDE after replacing all the necessary libraries by the ones who came with the Github download) both are configured with the new settings I received from the new bot, both start up and are configured with the right wifi settings.

Both units show in the browser, but also both keep trying to connect to MQTT which fails.

> 19:52:21 Attempting MQTT connection...
19:52:21 If this is taking more than expected, connect to the config panel on the ip: 192.168.0.101 to review the MQTT connection credentials.
19:52:21 failed, rc=-2

Groundstation status shows:
Name | PE2BZnr2
Version | 21021701
MQTT Server | CONNECTED
WiFi | CONNECTED
Radio | READY
Test Mode | DISABLED

Anyone a clue what is happening ?

Thanks

zanco, 2021-02-18:
It must be a library issue in the IDE.

I used the option to update the firmware from webbrowser and flashed both with the Firmware OTA bin file

Now both stations are connected to MQTT

4m1g0, 2021-02-21:
Thank you for the the follow up. The libraries were all updated so it's very important to copy them into the lib repo if you are using Arduino IDE.

I'm glad you could fix it!

closed 2021-02-21 by 4m1g0