tinygs/tinyGS #65: Display date next to Time in TinyGS device web dashboard

DroneMechanic, 2021-03-01:
As there is only time, if the dashboard is left open for days. The user loses context of which day the reception occurred.
It is a user experience issue.

4m1g0, 2021-03-01:
Hey. 

I'm not sure if it's a good idea in the local console because the  buffer is quite limited, so the more we print the less history there is... What I think it could be cool is a custom link to the specific station dasboard on tinyGS.com so you can navigate in one click and that can help clarify all the receptions with decodes and many more info.

We'll think about it.

PintarIot, 2021-03-02:
I see, its a platform limitation as its coming straight from the device. 
i guess you can do some shortform hack later, or the webpage itself has the date range instead of in the serial of the device.

But yeah, the real question: is there many users asking why the data is not displayed- when it actually was already displayed on a previous day?

DroneMechanic, 2021-03-02:
Yeah thinking more about it, on Nicer on the web - then if its all there, and can change the view from the current individual labels to list like in the dashboard - so can see all the low level techie data in one window scrolling up n down.

The local dashboard for troubleshooting purposes only.