tinygs/tinyGS #71: Mac uploader seems broken

skandragon, 2021-03-21:
When I run the Mac uploader, nothing seems to happen.  It shows as running in the dock, but no windows appear, nor does it accept standard "quit" and requires a "force quit" to terminate.

jasonfish, 2021-03-22:
Same issue here. I ended up using ESPHome-Flasher.

4m1g0, 2021-03-23:
Can you please check if you experience the same with tasmotizer? https://github.com/tasmota/tasmotizer

thanks

jasonfish, 2021-03-23:
Tosmotizer runs fine, but couldn't flash the bin. Message about "bad header" or similar.

4m1g0, 2021-03-23:
Thanks, I'll try to take a look to it. Looks like it only fails on the new versions of the OS. Problem is that I don't have a mac to test.

skandragon, 2021-03-29:
I ended up using the platformIO uploader, and it worked fine.

freilich, 2021-04-15:
Same experience here under macOS 11.  On macOS 10.15 (Catalina), the uploader launches and crashes during the "download" phase of the install.

`Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY`

GuidosLabs, 2021-06-12:
I'm experiencing the same issue with the uploader on Catalina. I saw the reply from skandragon, so I installed the platformIO extension on VSCode, cloned the repo, started VSCode, selected the board and serial port, and added the tinyGS root folder to the workspace. But I'm not able to build and upload to my device.  From the docs, it looks like the only thing I should have to do is click the right-arrow icon (Platform: Upload when I hover on it) on the task bar, but then I get a pull-down list at the top asking to "Select the task to run" and I don't know how to proceed. I tried selecting the highlighted entries, but no luck. Any advice?

pgcudahy, 2022-04-24:
Hi, this seems to be still broken. Has any progress been made? In my console I get several errors like 
`Failed to validate dynamic code signature of accessing process identifier=<ID of InvalidCode>, pid=1107, auid=501, euid=501, binary_path=/Applications/TinyGS_Uploader.app/Contents/MacOS/TinyGS_Uploader: #-67062: Error Domain=NSOSStatusErrorDomain Code=-67062 "(null)"`

`System Policy: TinyGS_Uploader(1108) deny(1) system-privilege 10006` and 

`Prompting policy for hardened runtime; service: kTCCServiceAppleEvents requires entitlement com.apple.security.automation.apple-events but it is missing for responsible={identifier=<ID of InvalidCode>, pid=1107, auid=501, euid=501, responsible_path=/Applications/TinyGS_Uploader.app/Contents/MacOS/TinyGS_Uploader, binary_path=/Applications/TinyGS_Uploader.app/Contents/MacOS/TinyGS_Uploader}, accessing={identifier=<ID of InvalidCode>, pid=1108, auid=501, euid=501, binary_path=/Applications/TinyGS_Uploader.app/Contents/MacOS/TinyGS_Uploader}, requesting={identifier=com.apple.appleeventsd, pid=435, auid=55, euid=55, binary_path=/System/Library/CoreServices/appleeventsd},`

Dzaring, 2022-11-01:
I got the same problem.  It starts, or it shows up in tray but nothing appears on the screen.  I have to force-quit it.

exxrxxe, 2022-12-03:
Same issue, shows in tray but is not accessible. macOS 13.0.1

zcrowman, 2023-11-11:
Several years later and the MacOS installer seems to still be broken. Really?

Dzaring, 2023-11-11:
What are you trying to install? Check internet connection. Sent from my iPhone

zcrowman, 2023-11-12:
I tried using the MacOS installer. I tried it on a 2009 iMac with Catalina; a 2014 Mac Mini on Big Sur; and a 2021 M1 Mac Mini on Sonoma. The installer crashes/freezes on every machine. So I tried it with Visual Studio and the recommended add-on. While it worked, it left my LoRa bricked and now nothing will work on it.  So this has been rather frustrating so far.

Dzaring, 2023-11-12:
You’re installing something from the App Store?Sent from my iPhone

zcrowman, 2024-01-04:
No. Not from the app store. The Mac firmware you provide on GitHub. It just doesn't seem to work and it seems it hasn't for several years. I ended up needing to go to a friend with a PC. Took like 5 minutes and it was flawless. Would be great if Mac users could have this same experience.

DroneMechanic, 2024-01-04:
I found it's usually the mac serial port drivers that screw up the uploading firmware. I have windows in bootcamp environment on the same laptop and the windows usb to serial drivers work fine and am able to upload no worries. Some hardware ttgo dont work well with macos usb serial port drivers becos of the chipset they use. Might need to find drivers. For myself i just managed to do it all in windows os

gmag11, 2026-10-03:
Thanks for the report! The desktop uploader is no longer part of the project; the supported way to flash a board for the first time is the web installer at https://installer.tinygs.com, as described in the README. Closing as obsolete - please reopen if you run into problems with the web installer.

closed 2026-10-03 by gmag11