tinygs/tinyGS #73: ESP32 dev board + SX127X

NDC-SC, 2021-03-22:
The TinyGS.ino sketch indicates that the ESP32 dev board + SX127X (Custom build, OLED optional) is supported.
However, in the drop down Board Type list of the Ground Station configuration, the SX127X is not listed as an option, only the "Custom ESP32 Wroom + SX126x" with three options.

Which Board Type should be selected when using an ESP32 and a SX1278 LoRa board ?

4m1g0, 2021-03-22:
SX126x is the module all Heltec and TTGO boards use, so if you plan to build one you can just use the same pinout and select that board on the dropdown.

But in general any ESP32 board is supported, you can just make a template by yourself as explained here: https://github.com/G4lile0/tinyGS/wiki/Board-Templates

If you make a template from a off-the-shelf board, you can open an issue for it to be included on the dropdown in a future release.

Cloolalang, 2021-03-22:
Im also trying to set this up.
I have ESP32 Dev board and AI thinker RA-02 433Mhz radio.
So yeah, no OLED I hope that works..

DroneMechanic, 2021-03-22:
The RA-02 board - leave the bussy pin as 0

v0l, 2021-03-23:
https://github.com/G4lile0/tinyGS/blob/master/tinyGS/src/Radio/Radio.cpp#L47

Maybe it has something to do with this?

I tried without the `aADDR` check and it still fails, even though ive already tested the same pin config with https://github.com/sandeepmistry/arduino-LoRa

Im using `LOLIN D32` with `RA-02`
```
15:29:11 [SX12xx] Initializing ... 
15:29:11 {"name":"[433] ESP32 Wroom SX1278","aADDR":0,"oSDA":0,"oSCL":0,"oRST":0,"pBut":0,"led":5,"radio":1,"lNSS":4,"lDIO0":17,"lDIO1":16,"lBUSSY":0,"lRST":5,"lMISO":19,"lMOSI":23,"lSCK":18,"lTCXOV":0.0}
15:29:11 failed, code -2
LoRa initialization failed. Please connect to 10.100.0.71 and make sure the board selected matches your hardware.
```

DroneMechanic, 2021-03-23:
Your rst pin cannot be zero. There are a minimum number of gpio to be connected of the lora module.
The spi pins: mosi, miso, 
Rst
Nss
Dio0
Dio1

This is my pinout as an example:

Ra-02 Ai Thinker Lora 433 
ESP32devkitV1

NSS - GPIO15   | DI0 - GPIO2
RST - GPIO9     | DI1 - GPIO5
MOSI - GPIO23 | MISO - GPIO19
SCK - GPIO18   |  Board LED GPIO25
VCC - 3.3V. | radio - 1

v0l, 2021-03-23:
> Your rst pin cannot be zero. There are a minimum number of gpio to be connected of the lora module.
> The spi pins: mosi, miso,
> Rst
> Nss
> Dio0
> Dio1
>
> This is my pinout as an example:
>
> Ra-02 Ai Thinker [...]

`oRST` is the reset for OLED which im not using, `lRST` is set correctly

DroneMechanic, 2021-03-23:
I had the same error when my Dio1 was not connected. 
[image]

Mine is a mess but it worked, i have 4 of these using the RA-02.

Your other lora pins all connected?
So in the gs webpage your radio is status is red?

Which files did you edit the code?

v0l, 2021-03-23:
@DroneMechanic https://github.com/G4lile0/tinyGS/blob/master/tinyGS/src/Radio/Radio.cpp#L79

DroneMechanic, 2021-03-23:
No not there.
If u wish to add your board to the local dropdown list. 
There are other files to edit. But do so at own risk.

Atm i see that the lora board is not getting initiated by the mcu.

Wiring
Board config not correct.
Bad lora module.
Just some things that can cause the lora module init error.

Can check here:

https://github.com/G4lile0/tinyGS/wiki/Board-templates

And here

https://github.com/G4lile0/tinyGS/wiki/Ground-Station-configuration

v0l, 2021-03-23:
Its possible that the pins are mapped differently by this dev board, in either case, i just added some code to allow the use of the default SPI port.. https://github.com/v0l/tinyGS/commit/ba6ce873cbf13e86839f7858743505bea55a9aa5 

Its working fine now.

dikkedimi, 2022-02-02:
Hi All,

been trying out an RFM9x (433Mhz) Adafruit lora module. Nothing fancy, should just be an SX126 if I am correct. The module works with [this](https://learn.adafruit.com/adafruit-rfm69hcw-and-rfm96-rfm95-rfm98-lora-packet-padio-breakouts/arduino-wiring) wiring and the accompanying test sketches. At least, it is able to initialize the module and send (and listen) for lora packets. I only have one module, so I can't really test wether the radio is indeed receiving. But, it should initialize.
```
23:33:01 TinyGS Version 2105260 -
23:33:01 [SX12xx] Initializing ...
23:33:07 failed, code -705
```
seems to indicate it does not.
I have tried a lot of different pins but in the end I settled for the pins that were actually mentioned in the documentation [here](https://github.com/G4lile0/tinyGS/wiki/Ground-Station-configuration#custom-esp32-wroom--sx126x-crystal)
```
{"name":"433 ESP32+RFM95","aADDR":60,"oSDA":0,"oSCL":0,"oRST":0,"pBut":0,"led":0,"radio":0,"lNSS":5,"lDIO0":2,"lDIO1":12,"lBUSSY":0,"lRST":14,"lMISO":19,"lMOSI":23,"lSCK":18,"lTCXOV":0.0}
```
I have also tried
```
{"name":"433 ESP32+RFM95","radio":0,"lNSS":5,"lDIO0":2,"lDIO1":12,"lBUSSY":0,"lRST":14,"lMISO":19,"lMOSI":23,"lSCK":18,"lTCXOV":0.0}
```
to no avail.

My setup is still on breadboard, could that be the issue? Try a different breadboard? I'd like to know if it's working before I create a lot of work for myself. I checked the wires for continuity, all seems well.

silwalasmin, 2023-01-23:
My board template is 
```
{"name":"[433] esp32 LoRa","aADDR":60,"oSDA":21,"oSCL":22,"oRST":0,"pBut":0,"led":0,"radio":1,"lNSS":5,"lDIO0":26,"lDIO1":35,"lDIO2":34,"lBUSSY":0,"lRST":14,"lMISO":19,"lMOSI":23,"lSCK":18,"lTCXOV":0.0} 
```
Before configuring my board template, the oled display worked. But after this configuration my oled display doesn't work at all

closed 2023-08-08 by G4lile0

rvismit, 2024-12-19:
I used RA-02 and getting one error Radio Status Error: -2. Someone please help.