tinygs/tinyGS #76: OLED w/o RST

sacculus, 2021-03-25:
Hi
I made custom station based on LOLIN32 board with 0.96" OLED display which doesn't have an RST pin. But OLED doesn't show anything, screen is black.
I'm suppose that I2C connection works and OLED initializing properly because display blinks short time with some picture at moment when board restarts.
Is it possible to force to work my kind of OLED display?

jhm776, 2021-03-27:
I'm using two different OLEDs that don't have a reset pin, both work OK. Does your display go blank after this screen? (This same issue showed up on my GS about a week ago, upgrading to newest TinyGS-master does not resolve this problem for me)
(is this some MQTT server issue???? - TinyGS code has not changed but this issue popped up)
Link to photo of my display  [https://imgur.com/gallery/VgvOwVe](url)

sacculus, 2021-04-19:
Hi
I flashed my GS again and display started to work normally. Strange behavior, but I suppose that it would be possible because bad wiring with breadboard.
Pls, close issue with not a bug.
Thanks!

closed 2021-04-19 by sacculus