tinygs/tinyGS #83: save Power - Solar Panel powerd

syswombat, 2021-03-30:
Hello
if the OLED is to 0 is it the same then to be able to complete switch off?
Would be nice to have a function to go into deep sleep if no satelite should arrive!

have a nice day
vinc

volodink, 2021-04-02:
Yeah, agree.

4m1g0, 2021-04-06:
Hello,

Setting the brightness to 0 will turn off the OLED and also draw less current. But the board will not enter in sleep mode. 

Ground stations are conceived like gateways in a LoRaWAN network so they are constantly listening and waiting for requests from the operator. Having them sleeping most of the time will make it not possible to connect to the local dashboard to monitor the status, and it will not be possible to manage your stations in real time through the tinygs.com console.

Some of the satellites that will be launched can relay messages and this allows for lora communication through satellite. However it is not practical to communicate 2 groundstations as they are already connected to the internet. Here we need a node device which is a battery powered sensor with only lora communication that can be in the middle of nowhere sending data to the satellite that is received by groundstations on the network. We have already made some prove proof of concept using the FossaSat-1 protocol.

What i've said so far if for the "standard" gs use case, there is another use case in which a groundStation is powered with solar and needs to save power. We have in the roadmap plans to add an add an option to configure a gs as low power device and make it sleep until the next pass. However this will disable all the real time operation features and needs a complete different handling of the firmware and the backend so we still need to finish some more urgent tasks before.

We'll keep track of this here ;)

syswombat, 2021-04-06:
@4m1g0 
ok i do understand and it make sense yes - so i need bigger Panel/Battery to power it.

closed 2021-04-06 by syswombat