tinygs/tinyGS #87: Mobile GS

catalinalb, 2021-04-02:
Hello,

Since I am using the TTGO/T-beam boards which have battery and GPS receiver, will be nice to auto-update the GPS position. 
Therefore creating a mobile Ground Station by attaching it to a car for example.

Thank you

syswombat, 2021-05-06:
and how to update your Board, or will your car be a AP station too?

catalinalb, 2021-05-06:
Actually I do carry an LTE modem (fritz.box) connected to a battery in my car. The antenna sits on top of the car on a magnetic mount, the roof acts as a ground plan. I do receive very well if I don't park the car very close to the house. 😉🙂

sfjuocekr, 2022-01-01:
I just had the same idea, my GS might be stationary but it could change location if I moved it. Updating the location via the GPS module might be preferable if the deviation from its set position is now in a different grid square?

For example I'm in JO33hf right now, but I might move to my brother over in JO22xf during the weekends and bring my toys over :)

waterwin, 2022-01-31:
A related question : is the Lora module / TinyGS station saving received packets when not connected to the internet or are these packets lost ?