tinygs/tinyGS #88: Linux uploader broken?

Mark-Wills, 2021-04-03:
Good morning,

I have today attempted to program my hardware using the firmware uploader program on Linux Mint V19.3 Cinnamon (64 bit). I downloaded file TinyGS_Uploader_LINUX.1 and made it executable with chmod +x

Upon execution, I receive the following error:

```./TinyGS_Uploader_LINUX.1 
[3844] Error loading Python lib '/tmp/_MEITWQals/libpython3.7m.so.1.0': dlopen: /lib/x86_64-linux-gnu/libm.so.6: version GLIBC_2.29' not found (required by /tmp/_MEITWQals/libpython3.7m.so.1.0)```

It would appear that a dependency is missing from the build?

I will attempt to use platform.io but I thought I would let you know. In the meantime, if you have any suggestions or I have simply done something wrong, let me know and I will re-confirm.

Regards

jhm776, 2021-04-05:
I breaks due to "version `GLIBC_2.29' not found". Does the same in Devuan(beowulf) & Debian(buster) which use glibc 2.28. (was it developed on current Ubuntu which seems to use 2.31 ??)

4m1g0, 2021-04-05:
Thank for the report, this is because it was built under a redhat based system. I think maybe it's better to release the python version for linux or do a little more work to make a package that supports redhat and debian based systems. I have to investigate about it.

Mark-Wills, 2021-04-05:
Will be happy to try a new build on 64-bit Linux Ubuntu derivatives when the time is right. Thanks for the response.
Mark

mjago, 2021-04-06:
Uploader works fine for me (this week) on Ubuntu Mate LTS:

`
ldd --version
ldd (Ubuntu GLIBC 2.31-0ubuntu9.2) 2.31
`

`
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal
`

gmag11, 2026-10-03:
Thanks for the report! The desktop uploader is no longer part of the project; the supported way to flash a board is the web installer at https://installer.tinygs.com, as described in the README. Closing as obsolete - please reopen if you run into problems with the web installer.

closed 2026-10-03 by gmag11