tinygs/tinyGS #91: TTGO Lora 433 search for 915MHz sattelites

henkiejan1, 2021-04-10:
Hi,

I have buyed a TTGO Lora 433MHz module because that should be the best for receiving satelites. I have uploaded the program with the 1 click uploader and looks like working ok.
But the strange thing is that it looks to 915MHz satelites. Gives CRC errors of course. I have uploaded again but always falls back on 915MHz. Of course do i have selected the 433MHz board in the configuration. 

Really strange. What goes wrong???

4m1g0, 2021-04-10:
On the list of boards (configured on the local dashboard) you can choose the version 433 or 915 of your board. 

The auto tune feature is a new feature and need to make some predictions of satellite passes ahead so If on the first time you create the station, you select a 915 board, the algorithm will chose 915 satellites for you even if you change the board type afterwards.

On the station console in tinygs.com there will be options to configure the auto tune and even change board type. But in the meantime I recommend creating a brand new ground station by editing your configuration locally and choosing a different name (making sure a 433 hardware board is selected)

henkiejan1, 2021-04-10:
Thank you for your anwser. I should try again. I thought i has really selected 433 after flashing but i am not sure i has made a mistake. I will try :)