tinygs/tinyGS #92: Please add support for station telemetry.

DaveVdE, 2021-04-11:
I'm building a solar powered ground station, and I'm looking to have some way to monitor the state of the power system, in particular:

- Voltage of the solar panel
- Current going into the charger
- Voltage of the battery
- Current going into the load (the ground station itself)

My initial thought was to add an extra ESP32 with some current monitors and voltage dividers, but wouldn't it be better if there was a way to build it into the tinygs app itself, and being able to monitor it from the web console? I thought about forking, but with the auto-upload I don't think that's feasible.

How to proceed?

swaxolez, 2021-04-11:
I'm doing the same as well.  I'm waiting for a small solar charger from adafruit to arrive before I put it all together.  In the meantime, I've been playing with deep sleep and using a single 18605 battery. Unfortunately, I'm having issues with my wifi extender and am waiting for a new one to be delivered.  The deep sleep cycling is working well and the battery consumption is not too bad.  I think it should be possible with appropriate solar panels connected to keep it going indefinitely.  Two things I still need to implement are brownout detection and a low voltage strategy. .i.e. being careful not to try and draw too much current with wifi and/or lora while the battery is in a low state to prevent a brownout.

I am currently monitoring the battery voltage from the console - that's easy enough.  Extra ADC channels should be possible to monitor the solar panel voltage and use some shunt resistors for measuring current.  

Currently, In order to wake keep my GS awake long enough to perform diagnostics, etc I need to gauge when it will be coming back online and try to refresh the console for when it does.  I programmed a console command to (d) to toggle deep sleep.  This whole procedure is a royal pain in the arse.  If I miss my window I generally need to wait another 5 minutes.  Not an ideal scenario.

Being able to send mqtt commands for the unit to receive when it wakes up is really the only way to go.  I've been looking at the mqtt code but am still learning to understand it better. I think what we need are either dedicated topics for GS telemetry such as those you've already indicated or at least some configurable "user" topics that can be used for whatever we desire such as voltage, current, etc.

Lastly, I think it might also be useful to port the code to non-wifi lora modules such as the heltec cubecell and have the mqtt packets relayed to a local lora network.  Not having to fire up wifi for communication would save a lot of power!

G4lile0, 2021-04-11:
We are successfully testing a quick low power mode hack that reduce the power consumption by 3, it wil lbe available in the release that we will issue in the following days.

https://twitter.com/G4lile0/status/1380865075003686915

Real low power version will require a lot of changes in the firmware and in the back-end, is on the roadmap but will take a a bit of time. 

Only LoRa GS and mobile GS are on the roadmap as well, but I don't think that we will have them in the near future.