tinygs/tinyGS #93: Export packets

dk3wn, 2021-04-14:
Is there a way to export the received packets from the database?
a) own packets
b) packets from a specific satellite

Any hint is very welcome.
Thanks, 73 
Mike DK3WN

4m1g0, 2021-04-15:
Hello,

Our system is not ready yet to support that. We are working on it to make sure the services are not overloaded. We need to change some core things to make this kind of queries more efficient. That's the reason we currently can only show the last data on the web and there is no pagination.

I hope we can have some updates on this soon, this is also a prerequisite for the programmatic API.

dk3wn, 2021-04-15:
Hi,

 

thanks for the response. I think we have another bug on the dashboard.

The NORBY packet is truncated. A few bytes are missing at the end. In the database the frame is correct (Hexview is also correct), but in the dashboard, it is too short.

 

Dashboard shows:

13:50:08 8effffffff0a0601c9401700000000f10f00001c40c66d0f0042524b204d57205645523a30325f31320000000000000e0000ec0700000002100008e80a8700180000000000000000000000008368e20188b86c006f203128260000000000ffff00000000000009040c0f0f0f0f8f0f000606bffcb260082d390e0c000c00004

Hexview database:

8EFFFFFFFF0A0601C93AA500000000F10F0000AA3A5F1C0E0042524B204D57205645523A30325F31320000000000000E0000EC07000000020C0008F30A8400A8050000000000000000000000BD9C9B01691C820007CF2F282E000000FFFF000000000000000007040C0F0F0F0F8F0F000204BFFCB2600A492D0E0C000C0000102006191B08040702006010F21D2F70

13:52:08 8effffffff0a0601c9401900000000f10f00001e403e6e0f0042524b204d57205645523a30325f31320000000000000e0000ec0700000002100008e30a87001800000000000000000000000072fe24025d194a00e720312827000000ffff00000000000000000a040c0f0f0f0f8f0f000706bffcb2600a44390e0c000c0000d

Hexview Database shows:

8EFFFFFFFF0A0601C9401900000000F10F00001E403E6E0F0042524B204D57205645523A30325F31320000000000000E0000EC0700000002100008E30A87001800000000000000000000000072FE24025D194A00E720312827000000FFFF00000000000000000A040C0F0F0F0F8F0F000706BFFCB2600A44390E0C000C0000D0140A184E08070B050060100420A031

I think this is just a limitation of the output width in the dashboard. Would be nice if you could fix that.

 

Thanks, 73

Mike

DK3WN

4m1g0, 2021-04-16:
That's a good catch. It gets truncated by the console buffer size... We have to print in different lines, this was not trivial.

The commit will be out in the next release on the next days.

Thanks!

syswombat, 2021-04-23:
Hello
maybe to not overload your server, it would be possible to get the information directly from the own GS.
just into the log file itself where all is written down or just store it in a file to access it without login too
have a nice day
vinc

dk3wn, 2021-04-29:
Hello Vinc et al,

 

I find this idea incredibly good. It is the easiest way to archive the received data - especially since the RX window is very limited.

 

I would also be very happy about a proper timestamp - so date + time. Currently, old messages get lost in the RX window and you have no reference to the current date anymore.

 

The interval for requesting a firmware update can be reduced. You know how often updates appear - so every 12 or 24 hours would be sufficient ;-)

 

I would be happy if all these points would be implemented.

 

Thanks, 73 Mike

DK3WN

 

Translated with www.DeepL.com/Translator <http://www.DeepL.com/Translator>  (free version)

4m1g0, 2021-04-30:
Hi,

This month we are focusing on an update on the TinyGS web to add many new features including pagination to see the full history, statistics, graphs, among many others.

We want to make the TinyGS web, an application where you have a dashboard with your data and you can view, analyze, filter, download, etc. all of it. There, is where we are putting our efforts. Remember that the initial goal of the project was to send the received packets to the telegram channel (https://t.me/tinyGS_Telemetry). And we have started to work on the web application only 2.5 months ago. Please, just a little patience. We are close ;)

Suggestions and pull requests for the web app are appreciated too: https://github.com/4m1g0/tinyGS-webapp

The serial console is a log for the system, not really a list of packet. We can try to clean it more as discussed here https://github.com/G4lile0/tinyGS/issues/99

dk3wn, 2021-05-07:
Hi,

it would be also good to have a complete UTC timestamp (date/time) in the station dashboard.

Thanks,
Mike

gmag11, 2026-10-03:
Thanks for the idea! CSV export of your own station's received packets is now available in the web app, so this request is already covered. Closing as completed.

closed 2026-10-03 by gmag11