tinygs/tinyGS #94: Error in board template loading

funkytorino, 2021-04-18:
Hello,

line 47 at Radio.cpp, which is transcripted below, was unabling me to initialize radio. The 'aDDR' entry is required according to documentation about custom board templates, but including that OR condition was making the init routine to return without initializing radio. I cant figure out why it is included?

`if (error.code() != DeserializationError::Ok || doc.containsKey("aADDR"))`

4m1g0, 2021-04-18:
Hello, it's a known issue. It is already fixed on the next release that will be out thankfully tomorrow.

if you want to install it yourself you can do it from this branch. https://github.com/G4lile0/tinyGS/tree/prepare_release

closed 2021-04-18 by 4m1g0

funkytorino, 2021-04-18:
Oh, I see. Makes sense know.

Cheers.