tinygs/tinyGS #96: Rescue mode button long press

DroneMechanic, 2021-04-22:
Custom Esp32devkit v1 + RA02 - the long press seems to be on all the time. After the fw update keeps cycling into the local AP mode. 
Which pin number is being tested as the user pinout?

closed 2021-04-22 by DroneMechanic

doismellburning, 2021-05-01:
@DroneMechanic You closed this without any commentary, but I find myself having the same issue (TTGO T-Beam) - anything you could share please?

doismellburning, 2021-05-01:
Actually I think 813c6c3621e7efa7f2c76d44d7d258dc3bf86552 (or something between then and 2104210) has fixed things for me

letBulochkin, 2022-07-17:
I am stuck with the same problem, using TTGO T-Beam 868/915MHz with OLED. After setting up wi-fi settings my board connects to mny wi-fi for several seconds, then "Rescue mode forced by button long press!" message appears in the logs, and board resets to its inner AP. Firmware version 2105262, so it seems commit sent by @doismellburning didn't fix it.

benjaminemhf, 2022-11-28:
I've had the same issue with various standalone ESP32 DevKits. It only occurred when they were connected to my PC via USB, and it forced the rescue mode every single time. Changing the Progbutton/pBut pin and grounding it didn't help. The solution was powering the ESP32 with any other source; via USB charger or via the 5 V/3.3 V pins directly. I think having it connected to the PC forces the ESP's "boot" mode on power-on and tinyGS somehow interprets that as a long press on pin 0.

a-kalmykov, 2023-03-13:
I found the following things related to **T-Beam 1.1 868/915Mhz (SX1276 + NEO-6M GPS + OLED)**:
1. The correct value for `PROG__BUTTON` is `38`
2. The `INPUT_PULLUP` setting should be disabled for this board except the case with sleep mode (according to Meshtastic firmware sources)

**How to fix:**
1. Replace `INPUT_PULLUP` with `INPUT` in `tinyGS/tinyGS.ino` on the line `163`:
```c
// make sure to call doLoop at least once before starting to use the configManager
configManager.doLoop();
pinMode (configManager.getBoardConfig().PROG__BUTTON, INPUT);  // <-- here
displayInit();
```
2. Rebuild and upload to the board
3. Use the following Board Template (the `PROG__BUTTON` value here is 5th in order):
`{0x3c, 21, 22, 16, 38, 22, 1, 18, 26, 33, 32, 14, 19, 27, 5, 0.0f, "868-915Mhz T-BEAM 1.1 + OLED - Alt Prg Btn"}`