tinygs/tinyGS #97: T-Beam V1.1 with SX1262 not supported

yoramrotbach, 2021-04-24:
Hi,
I have TTGO T-Beam V1.1 which has SX1262 LoRa module.
The module fails to initialize and isn't supported with the builtin templates.
To make a long story short, here is how to solve it (not elegant but it works):
The problem is with the radio.cpp assuming the module to be SX1268. When you change all references of SX1268 to SX1262 and adding the right template - it works

4m1g0, 2021-04-24:
Can you provide the template you are using?

yoramrotbach, 2021-04-25:
Hi,
This template works:
{"name":"[915] LilyGo_TBeam_V1.1 SX1262 A","aADDR":60,"oSDA":21,"oSCL":22,"oRST":16,"pBut":38,"led":4,"radio":0,"lNSS":18,"lDIO0":26,"lDIO1":33,"lBUSSY":32,"lRST":23,"lMISO":19,"lMOSI":27,"lSCK":5,"lTCXOV":1.6}

This is the template you have posted on Telegram. However a double quote was missing before the word led.

In addition, the template wasn't doing anything until I have uncommented the else part in radio.cpp
 if (ConfigManager::getInstance().getBoardConfig().L_SX127X)
      state = ((SX1278*)lora)->_mod->SPIsetRegValue((regValue>>8)&0x0F, regValue&0x0F, (regMask>>16)&0x0F, (regMask>>8)&0x0F, regMask&0x0F);
   // else
   //   state = ((SX1268*)lora)->_mod->SPIsetRegValue((regValue>>8)&0x0F, regValue&0x0F, (regMask>>16)&0x0F, (regMask>>8)&0x0F, regMask&0x0F);

This is after I've renamed all SX1268 to SX1262. These two chips have the following differences: frequency bands and power. this is why 915 MHz will never work if the code is hard coded to SX1268.

I hope this helps. Please let me know if you would like me to test newer firmware versions 

Yoram

4m1g0, 2021-04-25:
Hello, 

Thanks. The commented lines are commented on purpose and should not affect any current radio configurations. They are for advanced config but they were wrong for 126x chips so they were commented out to avoid problems until we can properly test them, so changing the module type should be the only modification needed.

The problem is the freq range of sx1268 which doesn't allow frequencies above 810.0 the patch implemented for sx127x was a hack so if you already could make it work we will wait to properly support every different module without the hack and hopefully it's ready for the next release.

syswombat, 2021-05-06:
i have a  TTGO T-Beam V1.1  too.
to select the t-Beam it did not work only with "433 Mhz TTGO LoRa 32 v2" Board  it work.

this
{"name":"[915] LilyGo_TBeam_V1.1 SX1262 A","aADDR":60,"oSDA":21,"oSCL":22,"oRST":16,"pBut":38,"led":4,"radio":0,"lNSS":18,"lDIO0":26,"lDIO1":33,"lBUSSY":32,"lRST":23,"lMISO":19,"lMOSI":27,"lSCK":5,"lTCXOV":1.6}

have to be copy Pasted to the,

Modem startup
or 
Board Template

and which Board type have to be selected?

gmag11, 2026-10-03:
Thanks for the report! SX1262/SX1268 modules are now supported and custom boards can be defined with board templates, so please try the appropriate profile. Closing as resolved.

closed 2026-10-03 by gmag11