tinygs/tinyGS #98: Auto tuning 433/915

HaJo10, 2021-04-26:
Hi,
Auto tuning is somewhat not understandable? (For  me)
A first improvement would be to add two additionel select options 
- all sattelites  (as now)
- 433 Mhz sats (band 1+2 of sx1272)
- 915 Mhz sats (band 3 of sx1272)

So it would be clear what to listen to and not dependent on selecting the correct board at start. Also one could change to other board without bringing auto tuning to struggle. Automatics in background are quite often a mess, espescially if you cannot switch off.

Greetz Hajo

4m1g0, 2021-04-26:
Hi,

Thanks, It's planned to add the 2 different options on the auto tune config as you said to override the initial settings, however just for reference while we can do that:

By definition, the band that a station will be able to use depends on the hardware. That's why it is assigned when a board is selected: If a station is created with a 433 board autotune will use 433 satellites and if the board is 915 it will use 915 satellites. 

Changing the freq once it's set is not yet supported, the workaround to change the frequency is to change the station name so it's considered as a new station.

HaJo10, 2021-04-26:
Understandable, thank's

4m1g0, 2021-04-26:
Let's keep it open for future reference until the functionality is added

syswombat, 2021-05-16:
Hello
but 915/868 are able to work together as one GS?
have a nice day
vinc

HaJo10, 2021-05-17:
Hello,
yes, of course. 868/915 is same band from the view of the sx127x chips. It's same antenna input pin.
sincerely
Hajo