tinygs/tinyGS #99: Feature: add configurable time interval to check for firmware updates.

funkytorino, 2021-04-26:
Hi there, 

seems to me a little awkward to check for a new firmware every hour. I was thinking on a new configuration option to specify how often a new firmware should be checked.

I don't know how resources are assigned, but I can work on it on my spare time. Just let me know.

Regards,
JM.

4m1g0, 2021-04-30:
Hi,

The station is in constant communication with the server, the updates are just a very small check. There is no benefit in checking less frequently, it's important with so many stations to have flexibility to choose when to update every station to distribute the load and also being able to retry relatively fast if it fails. If we have to wait 24 hours for every update there is a very small margin. 

Another argument is that some firmware errors may make the station unstable after certain time of uptime, so having frequent checks give us the ability to quickly launch updates if we detect such kind of errors.

I think the message log is what that is bothering some people and not the check itself, so I guess removing the log will be the solution...

funkytorino, 2021-04-30:
Fair enough. Thanks for clarify!

JM