tinygs/tinyGS #112: Update VR3X-A/B/C NORAD numbers

mahtin, 2021-06-01:
Packets have `0` in the `norad` field for **VR3X-A** (and maybe B & C?). Hence this search for valid numbers ...

```
$ curl -sSLR 'http://www.planet4589.org/space/log/satcat.txt' | egrep 'V-R3X' 
S047463 2021-006BC     Cora                              ?      V-R3X 1                  NASA/CMU     2021 Jan 24 In Earth orbit                2021 Jan 27 LLEO/S      95.25    527 x    535 x  97.51
S047467 2021-006BG     Petrie                            ?      V-R3X 2                  NASA/CMU     2021 Jan 24 In Earth orbit                2021 Jan 26 LLEO/S      95.26    528 x    535 x  97.52
S047524 2021-006DQ     Littlefoot                        ?      V-R3X 3                  NASA/CMU     2021 Jan 24 In Earth orbit                2021 Jan 27 LLEO/S      95.26    528 x    535 x  97.52
$ 
```

So they are `47463`, `47467`, and `47524`. However, I'm wondering if the `1`, `2`, and `3` are `A`, `B`, and `C` or `C`, `B`, and `A`. (see https://tinygs.com/satellites).

Sadly (once again) Celestrack does not have the names of the satellites added. ...
```
$ curl -sSLR 'https://www.celestrak.com/pub/satcat.txt ' |  egrep '47463|47467|47524'
2021-006BC   47463     OBJECT BC                 TBD    2021-01-24  AFETR                 95.3   97.5     538     524     N/A       
2021-006BG   47467     OBJECT BG                 TBD    2021-01-24  AFETR                 95.3   97.5     538     524     N/A       
2021-006DQ   47524     OBJECT DQ                 TBD    2021-01-24  AFETR                 95.3   97.5     537     524     N/A       
$
```

BTW: Why **VR3X-A** on TinyGS vs **V-R3X-A**? Inquiring minds want to know.

G4lile0, 2021-06-03:
Thanks Martin, 

We have fixed the NORAD numbers on the autotune function and updated all the old packets on the database.

Here you have the decoder from  Max Holliday:
https://github.com/maholli/VR3X-decoder/blob/main/gs_parse_standalone.py

VR3X-A Littlefoot start with  --> 33 3A
VR3X-B Petrie     start with  --> 33 3B
VR3X-C Cera      start with  --> 33 3C

The information coming from planet4589 doesn't look accurate, in any case as all the sat share the same radio configuration, the decoder use the second byte to properly identify each satellite.

mahtin, 2021-07-10:
Please note that VR3X-A entries in the packets API still returns `0` (vs. `47463`). It once did return values correctly. Sadly, that regressed backwards.