tinygs/tinyGS #113: Support for m5stack board with LoRa module

lotrdocker, 2021-06-05:
The m5stack board is not in the supported board list, but I tested and compiled, uploaded TinyGS, with PlatformIO, in my m5stack board, with module LoRa 433, crossing my fingers that it works :). Compiling and uploading are ok but the m5stack'screen stay black. I will keep searching for TinyGS works, but has anyone already tried ?

PATTOOo, 2021-07-09:
May be we need to check m5stack pinouts for connected display

4m1g0, 2021-07-16:
The display for M5Stack has a different technology so I think a new library is needed. It could be interesting if someone can investigate it.

Regarding the lora pinouts, there is no change the code, all pins can be configrued through board templates: https://github.com/G4lile0/tinyGS/wiki/Board-Templates

Can you share the valid board template for m5stack? it might be useful for other users

lotrdocker, 2021-07-17:
Hi all, thanks for reply. For m5stack, apparently, it's a Ra-01 module for LoRa. I found this thread about a Ra-02 module https://github.com/G4lile0/tinyGS/issues/73. I continue my research and tests.

hollie, 2022-07-23:
Hey @lotrdocker 

did you make any progress porting TinyGS to the m5stack? Would you care to share your code?

Thanks,
 Lieven

Pedritoprint, 2022-12-22:
Any news? Solution for M5Stack?
Thanks HB9CMI

Pedritoprint, 2022-12-24:
> Hi all, thanks for reply. For m5stack, apparently, it's a Ra-01 module for LoRa. I found this thread about a Ra-02 module #73. I continue my research and tests.

But today, i think it's a RA-02 Chip.

ifrew, 2024-03-30:
This issue is closed but I got it going with the M5stack and the Lora 868 module. Should work fine with the Lora 433 too. As others mentioned no graphic display as its not an I2C but an SPI display but since its a ground station the display is not really needed.  Here is the board template. The key is not to use the DIO1 pin as that is not connected on the lora board from M5stack

{"name":"[915] M5Stack LoRa ","aADDR" :60, "oSDA" :21, "oSCL": 22, "oRST" :0, "pBut" :0, "radio":1,"lNSS" :5, "lDIO0":36, "lDIO1" :0,"lBUSSY":0,"lRST":26,"lMISO" :19,"lMOSI" :23,"lSCK" :18,"lTCXOV" :0.0}