tinygs/tinyGS #114: Wiki Typos: FAQ

jimt, 2021-06-08:
> I'm new and I want to get in, what hardware should I buy? Fist choose the satellites that you want to receive.

Fist -> First

> What type of anntenna should I use for TinyGS LoRa is an awesome technology and can receive signals under the noise floor. The standard duck or coil antennas that usually come with Heltec or TTGO boards were proven to receive some packets form quite long distances. However, the conditions have to be good to be able to receive with such 

anntenna -> antenna
form -> from

4m1g0, 2021-06-10:
Hello, Thanks for the review!

Can you edit it here? https://github.com/4m1g0/tinygs-wiki/blob/master/FAQ.md

Just click on the edit button and then send the changes proposal, that way the contribution is registered.

[image]

jimt, 2021-06-11:
Done in 4m1g0/tinygs-wiki#5

Sorry, I had looked for the wiki under G4lile0.

closed 2021-06-11 by jimt