tinygs/tinyGS #116: API call for TLE data doesn't return fresh information

mahtin, 2021-06-20:
The present https://api.tinygs.com/v1/tinygs_supported.txt API call seems to not update all its listed satellites. Presently, the `Epoch` is old in some entries. It good for `Norbi` and `SDSat`; but not the others. Fresher data is available for the other satellites.

```
$ curl -sSRL https://api.tinygs.com/v1/tinygs_supported.txt | paste - - - | awk -F'       ' '{ satellite=$1;line1=$2;line2=$3; epoch=substr(line1,19,14); printf "%-20s %s\n",satellite,epoch; }' 
Norbi                21170.90132541
FEES                 21144.56510677
SDSat                21170.71862042
VR3X-A               21053.86202839
VR3X-B               21053.86075148
VR3X-C               21054.12407896
Sri Shakthi Sat      21093.52252237
$ 
```

This information should be kept up to date for many reasons.

mahtin, 2021-06-22:
This code would keep the TLE info up to date. Run daily'ish.
```
$ for norad in 46494 47721 48082 47463 47467 47524 47716
> do
> curl -sSRL https://celestrak.com/satcat/tle.php?CATNR=$norad
> done
NORBI
1 46494U 20068J   21172.56501227  .00001242  00000-0  94969-4 0  9994
2 46494  97.6937 110.0878 0019406  67.3645 292.9627 15.03628496 39953
SDSAT                   
1 47721U 21015W   21173.15289366 -.00000070  00000-0  00000+0 0  9991
2 47721  97.4640 247.4174 0009466 233.4347 126.6208 15.20928941 17298
FEES                    
1 48082U 21022AL  21173.37447305  .00000837  00000-0  60552-4 0  9996
2 48082  97.5591  75.5557 0014463 288.3781  71.5873 15.06628573 12321
OBJECT BC         
1 47463U 21006BC  21172.83581329 -.00008258  00000-0 -48043-3 0  9992
2 47463  97.4840 233.5918 0011916 106.7482   4.5985 15.11771176 22692
OBJECT BG               
1 47467U 21006BG  21172.82056242 -.00003588  00000-0 -20608-3 0  9994
2 47467  97.4822 233.5984 0011764 105.6166 317.8824 15.11843213 22691
OBJECT DQ
1 47524U 21006DQ  21173.27725362  .00001049  00000-0  64882-4 0  9995
2 47524  97.4844 234.0842 0011390 102.2835 325.9805 15.11925325 22761
OBJECT S                
1 47716U 21015S   21172.64811092 -.00000070  00000-0  00000+0 0  9995
2 47716  97.4489 246.8827 0009179 253.4903 106.5725 15.21544646 17122
$
```
You could clean up the names by adding the following ...
```
| sed -e s/OBJECT BC/VR3X-A/ -e s/OBJECT BG/VR3X-B/ -e s/OBJECT DQ/VR3X-C/ -e s/OBJECT S/Sri Shakthi Sat/
```
to that script above.

mahtin, 2021-06-22:
Additionally ... `SDSat` is NORAD number `47721`; however, there's also an entry for `Sri Shakthi Sat` exists with `47716` which after some digging is not actually a satellite (that TinyGS) cares about. 

Hence ...
```
Sri Shakthi Sat
1 47716U 21015T   21093.52252237 -.00000070  00000+0  00000+0 0  9993
2 47716  97.4585 168.5207 0013232 160.9461 199.2332 15.21085427  5090
```
... should be deleted from the TLE file.

mahtin, 2021-07-10:
The TLE's for **VR3X-A**, **VR3X-B**, and **VR3X-C** are still not being kept up to date. This means that  should the satellites actually come back to life; they will not be heard by TinyGS stations (like mine). Please update your scripts so this is kept valid.

Thanks.

andypiper, 2024-06-10:
Observation, [https://api.tinygs.com/v1/tinygs_supported.txt](https://api.tinygs.com/v1/tinygs_supported.txt) and [https://api.tinygs.com/v1/tles.txt](https://api.tinygs.com/v1/tles.txt) appear to be the same data. Curious about the purpose here.

OK1SLM, 2025-01-18:
HI,
Please update TLE data and satellites.
Tnx