tinygs/tinyGS #118: Sorry i'm a beginner : I don't sucess to program the TTGO

pjbellin, 2021-07-01:
Hello all, and many thanks for taking time to read my post and try to help me. 
I've try to flash my TTGO with the uploader but it block after downloding the files and i've the following message : "Failed to connect to ESP32: Timed out waiting for packet header"

I've download and install the driver before connecting the TTGO to my computer wich work on windows 10. 
Nothing happen when i connect the TTGO to an usb port of the computer, no usb device is reconize and no port com appears. I've only in my device panel the PORT COM 1 and that's all. 

Any one had this problem ? I would like tot sucess to install my station and have sucess with it ! 

Have all a very nice day and long life to ham radio ! 

73'S QRO  F4COH

DroneMechanic, 2021-07-01:
Hi there, try a different usb cable, or there might be some usb chipset drivers that you need to install.
Usually the ttgo should pop up some msg on insert cable.
If you using windows open up device manager and then look for the ports sectionand expand the list.
Then plug in the ttgo - if a device with a yellow triangle popsup then its a driver issue. If nothing then its a cable issue.

pjbellin, 2021-07-02:
Hi and thanks for your quick help ! 
I'll try this evening and say you if everything goes fine ! 
Thanks a lot for your help !

pjbellin, 2021-07-06:
Hi : 

Solved ! Just buy a USB cable witch do data and not only charge ! 
I 've tried 5 cable and finally buy a brand new one certified for data ! 

Thanks all !

closed 2022-04-07 by 4m1g0