tinygs/tinyGS #120: API call https://api.tinygs.com/v1/station/NAME@USERID/statistics returns date/time segments based on CET/CEST

mahtin, 2021-07-06:
Presently, a call to the `station` `statistics` API endpoint uses `22:00` vs `00:00` for start/end of day. Could the count be normalized to be UTC midnight to UTC midnight? It's as if there's a midnight cronjob running on a CET/CEST set server somewhere.

Hint: all servers should run **UTC** - but I digress

4m1g0, 2021-07-08:
Thanks.

The API is not yet available. Those endpoints are adhoc developments for the web (which needs the data like that for historical reasons right now) it's not intended to use as a publicAPI and it's discouraged to do so as they are not properly secured yet (they ban globally) and they will change in the near future as we develop it.

We plan to publish a stable API with documentation and proper quota per endpoint hopefully in the next weeks.

mahtin, 2021-07-10:
Hi @4m1g0, Yes, I realized it wasn't a public API. My main point was to point out the **midnight** European Timezone bucketing within the code. I hope this can be fixed. 

I can't wait for the official API publication.

closed 2021-07-10 by mahtin