tinygs/tinyGS #128: LILYGO TTGO ESP32-Paxcounter LoRa32 V2.1 1.6.1 Version 433 MHZ LoRa ESP-32

DE3JGA, 2021-10-02:
Hello all,

would this LoRA32 work with V2.1 for the TinyGS project?

https://www.amazon.de/TTGO-ESP32-Paxcounter-LoRa32-Version-Bluetooth/dp/B08CGSYM4G/ref=sr_1_1?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=TTGO+LoRa32+V2+433&qid=1633160576&sr=8-1

Thanks and 73 
Josef
DE3JGA

DE3JGA, 2021-10-07:
OK thank you for the information!
73 Josef
DE3JGA

DrHyperKALICH, 2022-01-16:
Anyway LoRa32 V2.1 1.6.1 works great!

Board template:

`{"name":"[433] TTGO LoRA 32 v2.1_1.6","aADDR":60,"oSDA":21,"oSCL":22,"oRST":16,"pBut":0,"led":25,"radio":1,"lNSS":18,"lDIO0":26,"lDIO1":33,"lBUSSY":0,"lRST":23,"lMISO":19,"lMOSI":27,"lSCK":5,"lTCXOV":0.0}`

gmag11, 2026-10-03:
Thanks for the question! This is the same board as the supported "433MHz TTGO LoRA 32 v2", so that profile works for it. Closing as answered.

closed 2026-10-03 by gmag11