tinygs/tinyGS #139: Satellite legal status

natsfr, 2022-01-27:
Hello,
Apparently you recently supported a satellite using Lora and Amateur Band.
Interestingly enough it seems you has been denied coordination by IARU.
How do you stand on that point ?

Secondly, you use LORA which is a proprietary modulation in the Amateur Band
it's not clear how legal it is.

Third point the main mission of the satellite seems commercial (and again the coordination
was denied by IARU) so why servicing them and helping them to violate the Amateur Band
restriction ?

Thanks a lot and best regards,

F4FXL, 2022-01-27:
Hi,

I am not involved in this project at all never even got interested in it, but I feel compelled  to step in as your second point is null.

As I wrote on twitter : it is not illegal to use proprietary/patented modulation  scheme or protocols on ham bands as long as the legal limits for power and bandwidth are observed and there is equipment widely available to decipher them. Yet depending on your licensing authority you might need a special authorization to operate such modulations/protocols.

darksidelemm, 2022-01-27:
Regardless of the use of LoRa in the amateur radio bands (I still think it's an incredibly wasteful use of spectrum - 125 kHz bandwidth for ~1-2kbit's is ridiculous), this particular satellite was *declined* coordination by the IARU as it is a commercial project - http://www.amsatuk.me.uk/iaru/declined_detail.php?serialnum=809

There should at the very least be a notice on the satellite's page on tinyGS that this satellite is using the amateur radio bands without authorisation, like there is on the SatNOGS page: https://db.satnogs.org/satellite/HWLH-9040-6011-7603-7570

F4FXL, 2022-01-28:
> I still think it's an incredibly wasteful use of spectrum - 125 kHz bandwidth for ~1-2kbit's is ridiculous
@darksidelemm This just shows how you have no clue about digital transmission. Digital transmission is a trade off between throughput and reliability. You cannot have high throughput and be reliable in a noisy environment. In LoRa the bandwidth isn't use for throughput but to make it decodable even with very low SINR !

natsfr, 2022-01-28:
I think your answer is rude but I'll try to answer in a constructive way to address the real problem here.

First, the question is not to know if Lora is good technically or not but the fact that patent and lack of implementation documentation can be seen as a problem when you used for HAM Project (I can easily argue that in the case of satellite you can use way more efficient modulation than LORA for the same job but that's not the point here).

Second, this satellite was denied the use of HAM Radio band and that should be a show stopper in all case. What happening here is that a company is making a commercial service and use HAM Band without having any coordination for that and that point alone is enough to not support it.

natsfr, 2022-01-28:
> In the strictest sense, 70cm is not an amateur radio band assignment, in other words a frequency assignment in which Hams are the primary user. On 70cm Hams are a secondary user, not primary, and there are other primary users of this frequency range. In other words, Hams are guests on 70cm frequency band.
> 
> Semantics are important. The satellite was denied coordination as part of the Amateur Radio service because it is primarily a commercial endeavor, not because of the frequency it was using.
> 
> Furthermore, 70cm is not the only frequency assignment in which Hams have a secondary user status, for example 902-928Mhz is another in which LoRa and LoRaWAN networks are found in North and South America.

That's an interesting point, but for France (I only know that case maybe the same for other) HAM Radio is primary use between 434 and 440MHz shared with other primary with a maximum of 20kHz bandwidth. This satellite if I'm not mistaken is emitting at 437.2MHz (I don't know the Lora mode used for BW).

F4FXL, 2022-01-28:
@natsfr I did not want to sound rude, yet your claim that "proprietary" stuff as nothing to do on ham bands is invalid and I explained why.
In France the bandwidth is not restricted on 70cm, see last page [here](https://www.entreprises.gouv.fr/files/files/directions_services/numerique/consultations-publiques/annexe1-radioamateurs.pdf). In the past hams were even doing television transmissions on 70cm !

I second you on the point that commercial stuff has to stay off the band, that's definitely the issue we are dealing with here. But trying to deal with it based on the claim that operating LoRa on 70cm is unlawful will lead you nowhere. If the guys were using AX25, or morse code or whatever it'll be the same issue, I think we can both agree on that.

natsfr, 2022-01-28:
> @natsfr I did not want to sound rude, yet your claim that "proprietary" stuff as nothing to do on ham bands is invalid and I explained why. 

On that point I don't agree but I must concede that nothing tell if it's ok or not and after lot of discussions these last year it seems it could be really country dependent.

> In France the bandwidth is not restricted on 70cm, see last page [here](https://www.entreprises.gouv.fr/files/files/directions_services/numerique/consultations-publiques/annexe1-radioamateurs.pdf). In the past hams were even doing television transmissions on 70cm !

Sure you're right I made a confusion between bandwidth allowed by the law (which is not limited over 220MHz) and what is "usually" used by ham community in ham band plan.

> I second you on the point that commercial stuff has to stay off the band, that's definitely the issue we are dealing with here. But trying to deal with it based on the claim that operating LoRa on [...]

I agree that here the main problem is the main mission and the main goal which I suspect is not paying and not waiting for an attribution in the right band. 
I don't agree with the LoRa conclusion you draw (as I said previously there is no clear conclusion in my opinion) my point of view is that in order to receive LoRa you have to use their chip or some reverse engineering based solution.
In the first case we don't know what happen in the chip (in the sense it's not documented) and also IN MY OPINION it seems to contradict the experimentation goal of HAM radio. In the second case, well it's more complicated my best guess is that in Europe you're pretty safe on the patent and IP law if you don't sell solution using that, but that could be the opposite in other country with stronger constraint on intellectual property.

(This debate is nothing new there are already debate about Fusion and DStar)
We can continue to discuss the LoRa case elsewhere if you feel inclined to.

Concerning this satellite the general conclusion drawn by IARU is the mission is commercial, and at least over some countries they are violating the HAM Radio band. If nothing can be done worldwide maybe geofencing would be a sort of solution.

closed 2022-07-08 by 4m1g0