tinygs/tinyGS #141: Support for TTGO variant

enbeee, 2022-02-13:
Hi, any way to make this board work? https://de.aliexpress.com/i/4000059678125.html
It says "T95_V1.1" on the board. Pinout seems to be the same as here:
#elif defined(LILYGO_T95_V1_0)

#define I2C_SDA                     21
#define I2C_SCL                     22
#define OLED_RST                    UNUSE_PIN

#define RADIO_SCLK_PIN              18
#define RADIO_MISO_PIN              19
#define RADIO_MOSI_PIN              23
#define RADIO_CS_PIN                5
#define RADIO_DI0_PIN               26
#define RADIO_RST_PIN               4
#define RADIO_DIO1_PIN              33
#define RADIO_DIO2_PIN              32
#define RADIO_BUSY_PIN              UNUSE_PIN

#define ADC_PIN                     35

I've tried to plug this into a board template, but I only get this error.

14:31:49 TinyGS Version 2105260 - 2105260
14:31:49 [SX12xx] Initializing ... 
14:31:49 failed, code -2

Thanks and best regards!

enbeee, 2022-02-14:
Thank you four the input, but I still get the error after "initializing".
I've tried the same pinout with a LoRa test sketch and it seemed to work.

JasonRJ, 2022-02-15:
In addition to the board template above, in the Board Type drop-down make sure that 433 MHz TTGO LoRA 32 v2 is selected then press Apply.  Wait 10 seconds then power-cycle the board.  Does the error continue after this step? If so, what is the error code?

enbeee, 2022-02-15:
Sadly it still gives the "failed: -2"/"Radio NOT READY" error.