tinygs/tinyGS #144: 2,4GHz GS Configuration

Leila49, 2022-03-06:
hey , I was wondering if the TinyGS community supports an ESP32+SX1280 LoRa tranceiver for tracking satellites on 2?4GHz

G4lile0, 2022-03-06:
Sure, it's already supported on the beta branch.
 
https://github.com/G4lile0/tinyGS/tree/beta

Leila49, 2022-03-06:
Okey,thanks. 
And for the board configuration , can we do it with the quick start , or we have to program it plz ?

G4lile0, 2023-08-08:
Here you have the board configuration, that iIm using for the ESP32+SX1280 :

`{"name":"[2.4] Multi SX1280","aADDR":60,"oSDA":21,"oSCL":22,"oRST":16,"pBut":0,"led":22,"radio":8,"lNSS":5,"lDIO0":26,"lDIO1":34,"lBUSSY":32,"lRST":14,"lMISO":19,"lMOSI":27,"lSCK":18,"lTCXOV":0.0}`

closed 2023-08-08 by G4lile0