tinygs/tinyGS #148: LILYGO TTGO T-Beam ESP32 915Mhz  but shows 433 on desktop

boterea, 2022-03-30:
I purchased a LILYG TTGO T-Beam ESP32 915Mhz v.1.1 from Banggood. After selecting 868-915Mhz TTGO LoRa 32 v1 from the drop down and installing the desktop still shows 433Mhz and after running most of last evening has found no sats. I've tried deleting my first ground station and starting new, but end up at the same place, desktop still shows 433Mhz. I hope someone can give me some suggestions.

boterea, 2022-03-30:
After 4 hours I went back and looked at the board, and to my surprise it now shows LoRa @ 917.50MHz. But still shows no packets received.

K4KDR, 2022-03-30:
Hello!  There are no LoRa satellites currently active on 915 MHz.

https://tinygs.com/satellites

boterea, 2022-03-31:
Tnx, I appreciate the feed back.

boterea, 2022-04-01:
Tnx!

closed 2022-04-07 by 4m1g0