tinygs/tinyGS #151: Feature request: Dashboard map card timestamp or filtering

jimt, 2022-04-07:
It would be helpful if the card that pops up when clicking on one of the green (or red) dots on the map view of a station dashboard included a date/timestamp.  That would allow me to better gauge effects of antenna changes graphically.  The card includes lots of information, but not the date/time of the observation.

(Alternatively, some sort of "slider" or other filter on the map view that allowed me to limit by start/end dates might be an even better graphical indicator of antenna changes.)

G4lile0, 2022-04-24:
Thanks for the feedback, first part done, it will be public available on the next release:
https://t.me/c/1448773154/60247
Second option is under investigation.

gmag11, 2026-10-03:
Thanks for the suggestion! Packet popups in the map now show a timestamp and filtering is available, so this is covered. Closing as completed.

closed 2026-10-03 by gmag11