tinygs/tinyGS #164: TinyGS-Client doesn't compile on latest version of Visual Studio Code/Platformio - incompatibility with ESP V4.00

stefanpi1, 2022-06-13:
After a fresh an clean installation of my Visual Studio Code, I couldn't compile the TinyGS-Clieant any more.
Finally I found, that the latest Version of VSC(1.68.0) and Platformio (Core 6.0.2·Home 3.4.1) comes with Espressif32 V4.00, which seems not to be compatible with the TinyGS-Client.
For a first workaround I installed the Espressif3.xx package, which solved the problem for now.
The error-message when compiling with ESP 4.00 is like "include/time.h not found":

4m1g0, 2022-07-08:
This was already fixed in the last versions of beta and master branches 🙂

yjdwbj, 2023-05-19:
This error still exists in latest version.

```sh
.pio/libdeps/heltec_wifi_lora_32/RadioLib/src/RadioLib.h:51:4: warning: #warning "God mode active, I hope it was intentional. Buckle up, lads." [-Wcpp]
   #warning "God mode active, I hope it was intentional. Buckle up, lads."
    ^~~~~~~
In file included from tinyGS/src/Radio/Radio.h:27,
                 from /home/michael/3TB-DISK/github/tinyGS/tinyGS/tinyGS.ino:76:
.pio/libdeps/heltec_wifi_lora_32/RadioLib/src/RadioLib.h:51:4: warning: #warning "God mode active, I hope it was intentional. Buckle up, lads." [-Wcpp]
   #warning "God mode active, I hope it was intentional. Buckle up, lads."
    ^~~~~~~
In file included from /home/michael/3TB-DISK/github/tinyGS/tinyGS/tinyGS.ino:79:
lib/ESPNtpClient/src/ESPNtpClient.h:29:10: fatal error: include/time.h: No such file or directory
 #include "include/time.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/heltec_wifi_lora_32/src/tinyGS.ino.cpp.o] Error 1
tinyGS/src/Radio/Radio.cpp: In constructor 'Radio::Radio()':
tinyGS/src/Radio/Radio.cpp:32:11: error: 'VSPI' was not declared in this scope
```