tinygs/tinyGS #172: Ham license and Tx

tkerby, 2022-07-03:
Suggesting in the config that a ham license allows for transmission (even test packets) is wrong. Most hams don't have any privileges for transmitting on satellite or weather balloon bands. Even in the case that they did, it's usually a condition of the license that users identify themselves with their callsign - something that could be added with CW transmission in the radio library but doesn't exist at the moment.

It may be that satellite operators have licensing that would allow them to transmit from a network of ground stations but that wouldn't typically include the station owner transmitting for test purposes.

I'd suggest that any transmission is done into a dummy load. This should still be enough for stations in the same location to pick up a test message from the tiny amount of leakage

kd7eir, 2022-07-03:
70 Centimeters (420-450 MHz), 400 MHz LoRa band being monitored (430 - 450 MHz)
33 Centimeters (902-928 MHz), 900 MHz LoRa band being monitored (ISM license free 915 MHz)

Hams absolutely have rights on these frequencies.

The issue of identifying is a valid issue, except for the license-free ISM band, but to say that "Most hams don't have any privileges for transmitting on satellite or weather balloon bands" is inane.

tkerby, 2022-07-03:
I'm mainly talking about the FossaSats here on 401.7MHZ along with the ident issues. I find that  the majority of the time, my stations are tuned to 401.7 due to the number of FossaSats out there. It's entirely possible to send a test packet on that band which in the UK doesn't have any ham access (as I believe is the case in other countries too). 

We may need more granularity on the transmit bands allowed to be able to set a band plan that a station may transmit in. And we need to add CW ident too (which would be fairly easy). I wouldn't suggest ident in the Lora message as it's somewhat obfuscated from the average ham in deciding it

tkerby, 2022-07-03:
Also, the UK and I believe EU has no 915/917Mhz access at all as 868Mhz is the ISM band here. It's certainly not a ham band

marc9990, 2023-06-14:
first where do you find the HAM/Callsign activation button and where to set your callsign - can't seem to find that

marc9990, 2023-06-14:
in Belgium is 868Mhz a open band for all to use

closed 2023-08-08 by G4lile0