tinygs/tinyGS #183: MQTT PW without Telegram?

df2jp, 2022-10-10:
Hello people

I just flashed the firmware and I am supposed to enter a MQTT user and PW. Is there a way to get this data without TELEGRAM?

I am not a messanger user and don't want to be...

Greetings
Joe/DF2JP

4m1g0, 2022-10-11:
Hello,

Currently Telegram is required as the system is quite dependant on it and it's also used to verify the users identity without asking for any personal data. It can be used through the Telegram web application though.

We have been planning to add an alternative authentication method, but it's something that probably won't be ready in the near future.

Regards

janvgils, 2023-07-11:
I wonder if this requirement is still the case, I am also not a Telegram user and don't want to be.
Is there an alternative?

gmag11, 2026-10-03:
Thanks for asking! MQTT credentials are now available directly in the web app after email login, without needing Telegram. Closing as completed.

closed 2026-10-03 by gmag11