tinygs/tinyGS #190: "Allow Automatic Tuning" Ignored

KevWal, 2023-01-07:
Running version v2105260 (release) I have "Allow Automatic Tuning" un-ticked, in /config, and I have "Modem startup" parameters defined of:

{"mode":"LoRa","freq":433.654,"bw":41.7,"sf":11,"cr":8,"sw":12,"pwr":0,"cl":0,"pl":8,"gain":0,"crc":true,"fldro":0,"sat":"Default"}

For additional logging I also have "Advanced parameters" set of:

{"dmode":9}

But my station seems to ignore the "Allow Automatic Tuning" setting, and accepts retuning commands from the MQTT connection:

```
14:39:41 [SX12x8] Starting to listen to Default
14:40:35 Received MQTT message: tinygs/1706663180/G7PMO/cmnd/begine : {"mode":"LoRa","freq":401.7,"bw":125,"sf":11,"cr":8,"sw":18,"pwr":5,"cl":120,"pl":8,"gain":0,"crc":true,"fldro":1,"sat":"FossaSat-2E11","NORAD":52750}
14:40:35 [SX12x8] Starting to listen to FossaSat-2E11

14:41:12 {"Vbat":0,"Mem":184896}
14:41:17 [SX12x8] Starting to listen to Default

14:42:35 Received MQTT message: tinygs/1706663180/G7PMO/cmnd/begine : {"mode":"LoRa","freq":401.7,"bw":125,"sf":11,"cr":8,"sw":18,"pwr":5,"cl":120,"pl":8,"gain":0,"crc":true,"fldro":1,"sat":"FossaSat-2E11","NORAD":52750}
14:42:35 [SX12x8] Starting to listen to FossaSat-2E11

14:47:36 Received MQTT message: tinygs/1706663180/G7PMO/cmnd/begine : {"mode":"LoRa","freq":401.7,"bw":125,"sf":11,"cr":8,"sw":18,"pwr":5,"cl":120,"pl":8,"gain":0,"crc":true,"fldro":1,"sat":"FossaSat-2E8","NORAD":99991}
14:47:36 [SX12x8] Starting to listen to FossaSat-2E8

15:03:36 Received MQTT message: tinygs/1706663180/G7PMO/cmnd/begine : {"mode":"LoRa","freq":433.775,"bw":125,"sf":12,"cr":5,"sw":18,"pwr":5,"cl":120,"pl":8,"gain":0,"crc":true,"fldro":1,"sat":"uPaPe","NORAD":99999,"filter":[2,0,36,36]}
15:03:36 [SX12x8] Starting to listen to uPaPe
```

KevWal, 2023-01-07:
And the current fix seems to be to use the "/weblogin" from your TinyGS Personal Bot and set Auto Tune to Disabled in this web dashboard.  This then sends a "remoteTune: 0" MQTT message which the device listens too, and then you can use the local "/config" dashboard to set the "Modem startup" parameters you want:

```
16:01:42 Received MQTT message: tinygs/1706663180/G7PMO/cmnd/remoteTune : 0
16:01:42 Set Remote Tune to OFF
16:01:42 Received MQTT message: tinygs/1706663180/G7PMO/cmnd/beginp : {"mode":"LoRa","sat":"ISM_433","NORAD":99999,"freq":433.3,"bw":125,"pl":8,"pwr":5,"sf":7,"cr":7,"sw":18,"crc":true,"fldro":0,"cl":120,"gain":0}
16:01:42 [SX12x8] Starting to listen to ISM_433
16:02:24 [SX12x8] Starting to listen to Default
```

Cheers
Kev

G4lile0, 2023-08-08:
Since long time we doesn't use this checkbox info from the board, to active or disable it, it's only possible using the tinygs.com  web dashboard

closed 2023-08-08 by G4lile0

KevWal, 2023-08-08:
But surely for the sake of users new and old the box should then be removed?

Cheers
Kev