tinygs/tinyGS #191: RSSI also while waiting for a packet

I0KPT, 2023-02-04:
Is there a way to read on one's Station Console the RSSI even in the absence of decoded packets? It could be very useful to continuously monitor the health of the receiving station, discover interference, LNA or antenna failure, etc.

sorenandersson, 2023-02-14:
If you have a SX127X lora board then you can access the current RSSI value at any time by reading lora module register 0x18. 0x18 is defined as SX127X_REG_RSSI_VALUE in radiolib.  try "int16_t rssi = -164 + radio.readRegister(SX127X_REG_RSSI_VALUE);" For more info see Semtech SX127x manual. The rssi value tends to fluctuate a little so I calculate a moving average and it works fine.

I0KPT, 2023-02-14:
> If you have a SX127X lora board then you can access the current RSSI value at any time by reading lora module register 0x18. 0x18 is defined as SX127X_REG_RSSI_VALUE in radiolib. try "int16_t rssi = [...]

Thanks, but I already successfully did, by an ad-hoc script, what you are today suggesting . What I was hoping is to read that register also during normal operation of TinyGS ground station, for example as an additional line in the station dashbord display. In other words, I suggest that new versions of ground station firmware will implement this functionality, too.

I0KPT, 2023-02-14:
Here you can find what I plotted using my had-hoc script in Arduino IDE:  (https://user-images.githubusercontent.com/50542629/218784073-ae0f6f72-2534-4ac6-b65d-b90e711c319e.jpg)

sorenandersson, 2023-02-14:
I have implemented RSSI and some other experimental functions on my TinyGs dashboard. You can see some screenshots in the Telegram TinyGs beta channel. Take a look at Oct 10, 2022. If you want to you can have a copy and try it out.

I0KPT, 2023-02-14:
> I have implemented RSSI and some other experimental functions on my TinyGs dashboard. You can see some screenshots in the Telegram TinyGs beta channel. Take a look at Oct 10, 2022. If you want to you can have a copy and try it out.

I went to Telegram TinyGS beta channel, but I can go no further back than January 23. Could you help me in locating your beta version?

sorenandersson, 2023-02-15:
I think you went to the "beta firmware test" channel inside "tinygs community". You need to make a global search for the "old" channel called "tinygs beta channel".

closed 2023-08-08 by G4lile0