tinygs/tinyGS #200: v2308041 - beta || boot loop w/ platform = espressif32 @ 6.3.2

K4KDR, 2023-08-04:
Hello!

Using the same TTGO ESP32 hardware that has been working since day-1 with previous tinyGS code (https://www.aliexpress.us/item/2251832729579512.html), I attempted to load the new BETA branch ( https://github.com/G4lile0/tinyGS/commit/1d776f5de3760af9a1bc1631276757d7426fe361 ) .

After upload, the serial monitor showed a continuous loop:

[image]

... after confirming that VSCode and related library / board / platforms were all up-to-date, I eventually found that the same BETA code works fine if the platform statement in **platformio.ini** was changed to the value that works w/ the version of code that I am currently running on other devices  ( `platform = espressif32@4.2.0` )

[image]

... and after upload, the Serial Monitor shows the expected menu:

[image]

Of course, newer versions of the expressif32 platform might also work - I did not try them all.

Also, I erased the ESP32 numerous times during testing but never got it to work with the 6.3.2 platform setting.

Thanks!

-Scott,  K4KDR

G4lile0, 2023-08-06:
Hi Scott, 

Thanks for the detailed info.. I guess that issue it's related to the OLED_RS allocated to the pin 16... and  connected to the FLASH_CS on your board.

 https://github.com/LilyGO/TTGO-LORA32/blob/master/schematic1in6.pdf
 [image]
 
 I just upload a new version that I hope it'll fix the problem. 
 
 Thanks

K4KDR, 2023-08-06:
No change, I'm afraid.

Using beta branch build https://github.com/G4lile0/tinyGS/commit/f1fc834ad7da72d16d04bf6404d44689091df00e , there was the same boot loop as written.    I also tried removing the **spaces** from the platform line in case that was an issue, but no help.

Only changing back to a previous version resulted in a normal boot

```
// platform = espressif32@6.3.2
platform = espressif32@4.2.0
```

FYI, the test device being used is:

https://tinygs.com/station/K4KDR_TTGO_433_4@1205281194

(I also changed the VERSION flag in `Status.h` to eliminate confusion here with uploading different builds)

[image]

G4lile0, 2023-08-06:
Hi Scott, 
I just upload a new commit.. seems you are using a ESP32-PICO-D4 so I disabled the auto config for your board.
Now, on the startup the chip model are shown .. please let me know if your chip it's the  ESP32-PICO-D4 or a different one.

[image]

Thanks

K4KDR, 2023-08-06:
Got out the magnifying glass....   yes, the marking on the chip for all these TTGO units (I have 5 or 6) is 'ESP32-PICO-D4'.     

I'll download the latest beta version & try again.

Thanks so much!

K4KDR, 2023-08-06:
Now booting normally - thanks for all the great work!

------------------------------------------------------------------------------
```
TinyGS Version 2308071 - 2105260-114-gf7c9b46
Chip  ESP32-PICO-D4 - 1
[SX12xx] Initializing ... 
[SX12x8] Starting to listen to Waiting
18:32:15 ------------- Controls -------------
18:32:15 e - erase board config and reset
18:32:15 b - reboot the board
18:32:15 p - send test packet to nearby stations (to check transmission)
18:32:15 ------------------------------------
17129
18:32:30 Attempting MQTT connection...
18:32:32 Connected to MQTT!
```

G4lile0, 2023-08-08:
Thanks for the feedback, it's fixed now so I close the issue.