tinygs/tinyGS #217: TinyGS Uploader not working on Ubuntu 22.04.3 LTS

OZ1SEJ, 2023-11-24:
I've downloaded the TinyGS uploader, but when I run it, I get this message:

```
$ ./TinyGS_Uploader_LINUX 
QSocketNotifier: Can only be used with threads started with QThread
qt.qpa.wayland: Failed to load client buffer integration: "wayland-egl"
qt.qpa.wayland: Available client buffer integrations: ()
qt.qpa.wayland: No shell integration named "xdg-shell" found
qt.qpa.wayland: No shell integration named "xdg-shell-v6" found
qt.qpa.wayland: No shell integration named "wl-shell" found
qt.qpa.wayland: No shell integration named "ivi-shell" found
qt.qpa.wayland: Loading shell integration failed.
qt.qpa.wayland: Attempted to load the following shells ("xdg-shell", "xdg-shell-v6", "wl-shell", "ivi-shell")
```

-and then it just hangs. It appears from the fine guide that I should see a window, but I don't. Am I doing something wrong, or am I missing something?

tdr112, 2023-11-30:
It does not work on Wayland. When logging in pick Xorg server, it will then work.

gmag11, 2026-10-03:
Thanks for the report! The desktop uploader is no longer part of the project; the supported way to flash a board is the web installer at https://installer.tinygs.com, as described in the README. Closing as obsolete - please reopen if you run into problems with the web installer.

closed 2026-10-03 by gmag11