tinygs/tinyGS #238: Heltec Wifi Lora32 v3 sx1262 no option for 868MHz

rodri16, 2024-04-15:
Hi I installed tinyGS to my Heltec lora32 v3 board sx1262 lora radio but with 868MHz
I configuration there is no option for 868MHz only 433MHz
[image]
Do I need to configured using templates?
Best

kb7sdm, 2024-05-08:
Hi there, I just installed on heltec v3 915mhz  and the auto detect in configuration was only either the heltec v3 433mhz or custom esp32 433mhz.. have you figured this out yet?

kb7sdm, 2024-05-08:
Thanks to some help in the telegram channel, I figured this out.  first ask your personal bot for your weblogin link, from the weblogin click on your station name, then select the green "operate" button and select Auto Tune Enabled 868-915 from the drop down and click save.

rodri16, 2024-05-09:
It's working Thanks @kb7sdm

rodri16, 2024-05-22:
I think there's still something that is not working
[image]

Dangermouse-UK, 2025-09-29:
Any updates on this? Heltec 868 seems unsupported?

gmag11, 2026-10-03:
Thanks for the question! The Heltec V3 profile covers 150-960 MHz and the frequency is set by the satellite, so a separate 868 MHz option is not needed. Closing as answered.

closed 2026-10-03 by gmag11