tinygs/tinyGS #255: How do you join the Telegraph group?

chichimus, 2024-07-31:
I cannot find any other way to contact anyone. Might be a dumb question but following the instructions it tells me to join a group on "telegraph". Go there, click on the "Join" and nothing happens?? Is this another app I have to add to a phone or what?

K4KDR, 2024-07-31:
Hello!

On the main page of this repository (https://github.com/G4lile0/tinyGS), scroll down to the section shown below for links to the various **Telegram** channels:

[image]

chichimus, 2024-08-01:
Ok I am not understanding something

it takes me to the join group page. Click on that all day an nothing happens.  Finally I downloaded telegraph and it now  wants to scan something from my phone? or give them my phone#  What has my phone got to do with any of this?