tinygs/tinyGS #256: Unable to connect HA mosquitto broker to tinyGS

pacmac, 2024-08-28:
Spent the last 4 days on and off trying to setup a bridge to tinygs so that I can integrate it, and everything I with the help ChatGPT has failed.

I have tried regenerating certs using multiple suggestions and nothing I do succeeds:

```
2024-08-28 15:45:25: Bridge local.core-mosquitto.tinygs doing local SUBSCRIBE on topic #
2024-08-28 15:45:25: Connecting bridge tinygs (mqtt.tinygs.com:1883)
2024-08-28 15:45:25: Warning: Unable to connect to bridge tinygs.
```

My bridge.conf file currently is:

```
connection tinygs
address mqtt.tinygs.com 8883
remote_username -1090701xxxx
remote_password 2vqPwi7m60xxxxx
topic # out 0
topic # in 0

# bridge_identity tinygs
# bridge_psk ef7a6c479b9d6dcf6c4530483b28e12270a50c4701b27d7e6a034f782595d046
# use_identity_as_username true
# cleansession true
# bridge_require_ocsp true

# TLS/SSL Configuration
# bridge_insecure true
# tls_version tlsv1.3
bridge_cafile /share/mosquitto/ssl/DSTroot_CA.crt
bridge_certfile /share/mosquitto/ssl/newRoot_CA.crt
bridge_keyfile /share/mosquitto/ssl/client.key
```

When I test openssl from command line, no matter what config I use I always get: "unable to get local issuer certificate"

```
SIDaUxdKNpExLyGCryjhsHGwmAvlOezZHLDVUrwxL8fPOdE1bVbxkA==
-----END CERTIFICATE-----
subject=C=ES, ST=MADRID, L=Madrid, O=TinyGS, OU=TinyGS, CN=mqtt.tinygs.com, emailAddress=tinygs@tinygs.com
issuer=C=ES, ST=MADRID, O=TinyGS, OU=TinyGS, CN=TinyGS Intermediary CA, emailAddress=tinygs@tinygs.com
---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: RSA-PSS
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 3158 bytes and written 403 bytes
Verification error: unable to get local issuer certificate
---
New, TLSv1.3, Cipher is TLS_AES_256_GCM_SHA384
Server public key is 2048 bit
This TLS version forbids renegotiation.
No ALPN negotiated
Early data was not sent
Verify return code: 20 (unable to get local issuer certificate)
---
```

I have tried using my own certs and also copied them out of certs.h no difference.

As you can see from the commented lines, I have tried multiple combinations of configs and nothing has worked.

I have also used suggestions from: http://www.steves-internet-guide.com/mosquitto-tls/ and nothing worked.

Really appreciate if you could provide a working ebridge xample and / or suggest how I can resolve it.

diginfo, 2024-10-28:
Bump;

Can someone please help with this ??

Thanks

hb9eue, 2025-02-22:
Just tried with MQTT-Explorer. Connection was only possible on port 8883 with TLS enabled. Port 1883 is not listening, and I had to disable certificate validation.

So basically if bridge_insecure true tells mosquitto to not validate the remote cert whe connecting in Bridge Mode, that should work.

You know what? I just go ahead and test. I'm already bridging Meshtastic and TTN to my local Mosquitto Broker, so why not one more :-)

Well I also fail and I might have an idea... is the project working with a private, non publicly signed root cert?

`mosquitto_sub -i tinyGS -h mqtt.tinygs.com -p 8883 --tls-use-os-certs --insecure -t '$SYS' -d
`
--insecure only remove the check for the cn= in the cert. I found no option to disable cert validation alltogether. So if the broker also is not able to validate the server cert, this is probably why the connection fails.

hb9eue, 2025-02-22:
```
# openssl s_client -verify 9 -CApath /etc/ssl/certs -connect mqtt.tinygs.com:8883
verify depth is 9
CONNECTED(00000003)
depth=1 C = ES, ST = MADRID, O = TinyGS, OU = TinyGS, CN = TinyGS Intermediary CA, emailAddress = tinygs@tinygs.com
verify error:num=20:unable to get local issuer certificate
verify return:1
depth=0 C = ES, ST = MADRID, L = Madrid, O = TinyGS, OU = TinyGS, CN = mqtt.tinygs.com, emailAddress = tinygs@tinygs.com
verify return:1
---
Certificate chain
 0 s:C = ES, ST = MADRID, L = Madrid, O = TinyGS, OU = TinyGS, CN = mqtt.tinygs.com, emailAddress = tinygs@tinygs.com
   i:C = ES, ST = MADRID, O = TinyGS, OU = TinyGS, CN = TinyGS Intermediary CA, emailAddress = tinygs@tinygs.com
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA384
   v:NotBefore: Jun 26 16:24:33 2024 GMT; NotAfter: Jun 21 16:24:33 2025 GMT
 1 s:C = ES, ST = MADRID, O = TinyGS, OU = TinyGS, CN = TinyGS Intermediary CA, emailAddress = tinygs@tinygs.com
   i:C = ES, ST = MADRID, L = Madrid, O = TinyGS, OU = TinyGS, CN = TinyGS Root CA, emailAddress = tinygs@tinygs.com
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: May 29 22:43:07 2022 GMT; NotAfter: May 26 22:43:07 2032 GMT
```

If we can find the public root cert (in the source-tryy maybee), we should be able to get the client to be able to validate it.

hb9eue, 2025-02-22:
Ok, I see from your first post, you already found:

https://github.com/G4lile0/tinyGS/blob/master/tinyGS/src/certs.h

hb9eue, 2025-02-22:
I just copied both certs from certs.h into a file tinyGS.crt

```
# mosquitto_sub -i tinyGS -h mqtt.tinygs.com -p 8883 -u 9434***** -P ***** --cafile /tmp/tinyGS.crt --insecure -t '$SYS' -d
Client tinyGS sending CONNECT
Client tinyGS received CONNACK (0)
Client tinyGS sending SUBSCRIBE (Mid: 1, Topic: $SYS, QoS: 0, Options: 0x00)
Client tinyGS received SUBACK
Subscribed (mid: 1): 0
```

success! Now let's try via broker.

hb9eue, 2025-02-22:
/etc/mosquitto/conf.d/tinyGS.conf

```
connection_messages true
log_type all

connection tingGS
address mqtt.tinygs.com:8883
remote_username 9434*****
remote_password *****
topic # both 0 tinygs/ /
bridge_cafile /tmp/tinyGS.crt
```

 Log looks good:

```
1740240292: Bridge local.pine64.tingGS doing local SUBSCRIBE on topic tinygs/#
1740240292: Connecting bridge (step 1) tingGS (mqtt.tinygs.com:8883)
1740240292: mosquitto version 2.0.18 running
1740240292: Connecting bridge (step 2) tingGS (mqtt.tinygs.com:8883)
1740240292: Bridge pine64.tingGS sending CONNECT
1740240292: Received CONNACK on connection local.pine64.tingGS.
1740240292: Bridge local.pine64.tingGS sending SUBSCRIBE (Mid: 4, Topic: /#, QoS: 0, Options: 0x00)
1740240292: Received PUBACK from local.pine64.tingGS (Mid: 3, RC:0)
1740240292: Received SUBACK from local.pine64.tingGS
```

You probably need to tweak the topic mapping. Not sure if it works this way.

Karol-NMD, 2025-10-03:
> I just copied both certs from certs.h into a file tinyGS.crt
> 
> ```
> # mosquitto_sub -i tinyGS -h mqtt.tinygs.com -p 8883 -u 9434***** -P ***** --cafile /tmp/tinyGS.crt -t '$SYS' -d
> Client tinyGS sending CONNECT
> Client tinyGS received CONNACK (0)
> Client tinyGS sending SUBSCRIBE (Mid: 1, Topic: $SYS, QoS: 0, Options: 0x00)
> Client tinyGS received SUBACK
> Subscribed (mid: 1): 0
> ```
> 
> success, even with cn validation! Now let's try via broker.

Hey, I'm trying to understand what you did here. But before, let me explain what I'm trying to do. I wish to locally receive the MQTT messages from my TinyGS on my MQTT Server (Raspberry pi). What you are proposing is to bridge the connection between the TinyGS server and my Rasp Pi?

pdxlocations, 2026-05-25:
> I just copied both certs from certs.h into a file tinyGS.crt

Could you elaborate on this a bit?  I'd like to look at the mqtt stream with MQTT Explorer but I'm not able to connect.