tinygs/tinyGS #259: REST API

sincos-pi, 2024-10-02:
Good afternoon, can I use the REST API to interact with your server, for example, to get data about one spacecraft and one satellite grouping? That would be very interesting.

darksidelemm, 2025-03-15:
I note that some packets can be extracted using a URL like: https://api.tinygs.com/v2/packets?satellite=DJ2DS-L2
where the 'packets' entry in the returned JSON object contains a list of objects with packet data.

e.g.:
```
{"raw":"JCRESjJEUy1MMiw0LDE2OjUxOjMxLDM4LjIyMjk4LC0wLjU1ODYyLDEwNzE0LDI0LDIzMywtMzYuMCwzLjYzLDAxMjM0NTY4QTkqMEZFMQo=","serverTime":1736873500552,"mode":"LoRa","freq":432.662,"sf":8,"bw":62.5,"cr":5,"bitrate":null,"freqDev":null,"rxBw":null,"satellite":"DJ2DS-L2","norad":99999,"id":"f14cbf3d-ab37-41cb-ab86-b36511f589ad","satDisplayName":"DJ2DS-L2","parsed":{"payload":{"magic":[36,36],"callsing":"DJ2DS-L2","pckCount":"4","pckTime":"16:51:31","tinygsLat":"38.22298","tinygsLng":"-0.55862","tinygsAlt":"10714","tinygsGPS":"24","tinygsHeading":"233","tinygsTemp":"-36.0","tinygsMainVoltage":"3.63","tinygsSolarVoltage":"01234568A9","tinygsCRC":"0FE1\n","tinygsTxpower":10},"telemetry":true},"stationNumber":1},
```

I have some interest in this in order to extract balloon telemetry for upload to SondeHub-Amateur. 

Some questions:
- Is use of this URL to extract packets supported and allowed?
- What other parameters can be used? e.g. to perhaps request only packets received within the last X days.
- I note that the 'callsign' field has been typo'd as 'callsing' - hopefully this is not intentional and will be fixed?

It also looks like metadata for individual packets can be downloaded via https://api.tinygs.com/v1/packet/f14cbf3d-ab37-41cb-ab86-b36511f589ad

G4lile0, 2025-03-16:
Regarding your questions, yes, the URL you mentioned can be used to extract packets. However, please be aware that we have very limited hardware resources, and our API usage is currently quite restrictive to avoid impacting the network.

Will be great to send the  telemetry to SondeHub-Amateur. To avoid putting extra load on our database, this would ideally need to be done in real-time, similar to how we handle data for satellite operators.

Thank you for pointing out the typo in the 'callsing' field! I will correct the parser so that new packets will have the correct 'callsign' field.

darksidelemm, 2025-03-17:
Yes, a live feed (via websockets, MQTT, or something similar) would be the best approach here. Polling your API regularly is not a suitable solution.