tinygs/tinyGS #268: Add new satellite: MTUSI-1 (438.100 MHz, LoRa)

MrIvaha, 2025-03-31:
### MTUSI-1 (61750)

### Transmission Parameters  
- Frequency: 438.100 MHz
- Modulation: LoRa
- Protocol: Raw binary

### Sample Data
```hex
0000: 42 a2 d1 00 00 01 01 66 30 20 6d 61 69 6e 20 61  B.....f0 main a
0010: 70 70 20 68 77 56 65 72 73 69 6f 6e 3a 20 30 20  pp hwVersion: 0 
0020: 66 77 56 65 72 73 69 6f 6e 3a 20 31 2e 31 20 64  fwVersion: 1.1 d
0030: 61 74 65 3a 20 4e 6f 76 20 31 33 20 32 30 32 33  ate: Nov 13 2023
0040: 20 31 39 3a 30 31 3a 35 35 00 00 00 00 00 00 00   19:01:55.......
0050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0080: 00 00 00 00 00 00 00 00 00 00 00 00 62 28 ff 2e  ............b(..
0090: 13 3a 00 00 00 5d 00 00 00 00 00 00 00 45 00 00  .:...].......E..
00a0: 00 0c cb 00 00 b7 3f ef 2e 5e 28 ff 2e 00 00 00  ......?..^(.....
00b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
```

### Additional Information
- NORAD ID: 61750
- Launch date: 2024-11-04T23:18:40+00:00
- Information: https://db.satnogs.org/satellite/LNIX-1421-5820-7502-3419

### Decoder Needed?
Yes (custom format, requires a Python script)

### Attachments

[image]

K4KDR, 2025-04-01:
UPDATE:

I believe that the sample from this morning's recording that I was using was RIGHT on the borderline of having enough SNR to decode.   Additionally, the Arduino UNO that I was using for testing various LoRa parameters kept crashing when an over-the-air test of that packet (via a PLUTO device) was received.    So, I moved to a much more powerful microcontroller (**STM32WL55JC2**) and with minor adjustments to my TX freq  (to squeeze every possible amount of signal from my I/Q recording), we do indeed have a valid decode!

[image]

... with the following settings in my STM32WL55JC2:  (disregard the TX-related LoRa parameters... my sketch also has TX capability for satellites that are configured for uplinks from the public)

[image]

K4KDR, 2025-04-02:
Since we are now receiving and decoding the HEX payloads, can you please share the python script that was mentioned in the original post?    Thanks!